moveit/moveit #1526: TrajectoryMonitor leaks memory

jschleicher, 2019-06-28:
### Description

Monitoring the memory usage of move_group via `htop`, multiple "Plan and execute" requests consume additional memory.

### Your environment
* ROS Distro: melodic
* Ubuntu 18.04
* Source build 898cdc8d8df5689319cd462c2c674954c6134604

### Steps to reproduce
1. `roslaunch panda_moveit_config demo.launch`
2. In RViz choose a random goal and press "Plan and Execute" (continue this step and watch memory consumption of the move_group process).

### Expected behaviour
Memory consumption stays constant after some initial runs.

### Actual behaviour
Memory footprint increases. Digging deeper:
https://github.com/ros-planning/moveit/blob/cfb46fe3ba6b0d708f22c8692c337a47d6cde6da/moveit_ros/planning/planning_scene_monitor/src/trajectory_monitor.cpp#L90
is never called. Adding a debug output to line 75 shows, that the trajectory gets more and more point from request to request.
```
ROS_INFO_STREAM("trajectory size is " << trajectory_.getWayPointCount());
```

Note:
* Memory consumption stays constant, if pressing "Plan" and "Execute" buttons as individual steps
* Only some kilobytes are allocated after some minutes running the robot (tiny increase), but over the long run this is still a problem for long-running applications, since it is never freed.

Question:
* What is the use case of `TrajectoryMonitor`? I cannot find, where the `trajectory_` is actually read / returned to the user.

rhaschke, 2019-06-28:
Thanks for point out this issue. Looks like TrajectoryMonitor is only used in [`PlanExecution`](https://github.com/ros-planning/moveit/blob/91f55a58007df4e1ed2e8cff8393337f4f9a3596/moveit_ros/planning/plan_execution/include/moveit/plan_execution/plan_execution.h#L155), which calls [`startTrajectoryMonitor`](https://github.com/ros-planning/moveit/blob/91f55a58007df4e1ed2e8cff8393337f4f9a3596/moveit_ros/planning/plan_execution/src/plan_execution.cpp#L399) and [`stopTrajectoryMonitor`](https://github.com/ros-planning/moveit/blob/91f55a58007df4e1ed2e8cff8393337f4f9a3596/moveit_ros/planning/plan_execution/src/plan_execution.cpp#L444). However, aside from these calls, the recorded trajectory is never used. So, I don't have an answer to your question as well. @davetcoleman, @v4hn Do you have an idea?
My suggestion is to remove it in `master`, maybe also in melodic?

rhaschke, 2019-06-28:
The `TrajectoryMonitor` was introduced in 040a483a1 and inserted into `PlanExecution` in 6ea3a5ff43. @isucan, do you remember the purpose of this instance in `PlanExecution`?

jschleicher, 2019-07-08:
Fixed in #1538.

closed 2019-07-08 by jschleicher