moveit/moveit #1527: Modularize Setup Assistant

rhaschke, 2019-06-28:
In #1375, @MohmadAyman did a great initial work to modularize MSA. However, as pointed out in https://github.com/ros-planning/moveit/pull/1375#pullrequestreview-212596707 and https://github.com/ros-planning/moveit/pull/1375#issuecomment-480944057 there are several open TODOs to finish this attempt and merge the branch [`feature-modular-msa`](https://github.com/ros-planning/moveit/tree/feature-modular-msa) into master:
* Data modularization: Each widget should store its data locally. Currently all the data is stored centrally in `MoveItConfigData`.
* Add loading of existing config files to allow proper editing of files. This will render the `ConfigurationFilesWidget` obsolete as there is no danger to overwrite manual changes.
* Use Qt's UI files to layout widgets - instead of manually creating and layouting items.

MohmadAyman, 2019-08-25:
> Use Qt's UI files to layout widgets - instead of manually creating and layouting items.

Hey @rhaschke , sorry for the late reply.
Are we planning to do that for all the existing widgets? Do you think it is necessary for modularizing the MSA?

rhaschke, 2019-08-25:
It's not necessary, but a great improvement that will facilitate modifying the GUI layouts in the future.

MohmadAyman, 2019-08-25:
Would it be okay to add it to a separate issue maybe, or just not making it
a prerequisite to merge into master?

I believe the bigger the issue the less motivating it is to work on it.

rhaschke, 2019-08-26:
Sure.

MohmadAyman, 2019-10-20:
Hey @rhaschke , Do you think there is a way to rebase this to work with melodic without having to solve huge numbers of merge conflicts? so I wouldnt have to install ubuntu 16.

rhaschke, 2019-10-20:
Done. Rebasing onto melodic-devel was done without any conflicts. Do you prefer a rebase to melodic or to master? The latter I already prepared but didn't yet push.

MohmadAyman, 2019-10-21:
Thank you! the latter should work fine!

rhaschke, 2019-10-21:
Could you please clarify: You do prefer master over melodic-devel?

MohmadAyman, 2019-10-21:
Okay, on melodic-devel would be nice, as long as it is ros melodic my
problem is solved, thanks!

rhaschke, 2019-10-21:
If you just need ROS Melodic, I suggest rebasing onto master, which should be the target branch for this contribution anyways.

rhaschke, 2019-10-21:
Done. Travis is working on it: https://travis-ci.org/ros-planning/moveit/builds/600723588
Note that there are some issues reported by Travis (due to stronger error checking).

davetcoleman, 2020-08-19:
@MohmadAyman I'm noticing the [Modular MSA branch](https://github.com/ros-planning/moveit/tree/feature-modular-msa) is 14 months stale, any chance we could finish that effort up?

MohmadAyman, 2020-09-03:
Yes, sadly I couldn't realize an implementation that would work with the last set of requirements we've talked about with @rhaschke