moveit/moveit #1535: Failed to initialize SBPL Planning Context

matteoselvatici, 2019-07-04:
### Description

Configuration of Moveit! plugin for SBPL.
Once a random goal is generated from the Moveit! GUI of rviz and the plan request is done, it fails.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
*  Moveit! Binary build

### Steps to reproduce
The compiled workspace contains the components suggested in http://sbpl.net/Software:
-sbpl
-sbpl_geometry_utils
-smpl
-leatherman
-moveit_planners_sbpl (even if no more required, as https://github.com/aurone/moveit_planners_sbpl/issues/3 says ).
Robot config:
-pr2_common
-pr2_moveit_config
-pr2_sbpl_interface_config

The source code: https://drive.google.com/open?id=1f7qmmAzgi8zECbhWFrMSW5JKSHGj5XRM

For testing, I launched:
**roslaunch yumi_sbpl_interface_config demo.launch**

### Expected behaviour

Moveit! should see the SBPL planner config.

It is weird since `rosparam list` returns all the parameters.
It seem like it does not points to the parameters set in `pr2_sbpl_interface_config/config/sbpl_planning.yaml`
### Actual behaviour

While it actually does not, as reported in the Console output.

Am I missing any config/buinding ?

### Backtrace or Console output

[ INFO] [1562229334.760440827]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1562229334.760739121]: Planning attempt 1 of at most 1
[ WARN] [1562229334.761652434]: **No configuration found for the 'arastar_bfs_manip' algorithm**
[ INFO] [1562229334.761894482]: Created SBPL Robot Model for group 'right_arm'
[ INFO] [1562229334.761951416]: Set planning link to 'r_wrist_roll_link'
**[ERROR]** [1562229334.762113761]: **Missing parameter 'search'**
**[ERROR]** [1562229334.762158339]: **Failed to initialize SBPL Planning Context**

gavanderhoorn, 2019-07-04:
This appears to be a cross-post of [Failed to initialize SBPL Planning Context](http://answers.ros.org/question/327674).

wi0195ld, 2019-07-04:
> This appears to be a cross-post of [Failed to initialize SBPL Planning Context](http://answers.ros.org/question/327674).
Yes!