moveit/moveit #1536: remove dependency on eigenpy in master

v4hn, 2019-07-04:
Currently, [moveit_ros_planning_interface depends on eigenpy](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/src/wrap_python_move_group.cpp#L50).
I would like to see this dependency removed again soon, although I do not provide a solution here.

https://github.com/ros-planning/moveit/pull/1518 already provided a replacement for it, but `boost_numpy` did not exist back in xenial, so this is no solution as long as we want to stay compatible with Ubuntu 16.04.

To be precise, we depend on @wxmerkt's [eigen_catkin](https://github.com/ipab-slmc/eigenpy_catkin-release/tree/release/melodic/eigenpy) package which wraps eigenpy as an `ExternalProject` for catkin, although that is [not the correct/preferred way to handle third-party packages](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty). In the process it adds a cmake config which [the upstream package](https://github.com/stack-of-tasks/eigenpy) does not provide. As we [`find_package` eigenpy as a catkin COMPONENT](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/CMakeLists.txt#L21), that makes it impossible to use eigenpy installed from upstream (or even use upstream's non-standard PPA builds).
Instead upstream relies on pkg-config, but incidentally their pkgconfig file is broken too - it uses `-isystem` to include numpy which `pkg-config --cflags-only-I eigenpy` fails to extract.
So the build fails to work if we switch to `pkg_check_modules`.

Importantly, the catkin package [does not support a python3 build](https://github.com/ipab-slmc/eigenpy_catkin-release/blob/release/melodic/eigenpy/CMakeLists.txt#L21) and upstream also needs a patch to enable python3 with a newer boost version.
I managed to hack the package [to build with both](https://github.com/v4hn/eigenpy_catkin/tree/python3-boost1.69-fix-numpy-include), but the patches do not generalize beyond my setup and the whole build process of the library involving `ExternalProject` and a [git submodule with a whole cmake subsystem](https://github.com/stack-of-tasks/eigenpy/blob/master/CMakeLists.txt) to detect standard dependencies is rather fragile.

wxmerkt, 2019-07-04:
> although that is not the correct/preferred way to handle third-party packages

I agree it is not the correct way of doing it - and the reason is that eigenpy and other CNRS projects force-install Python into `site-packages` (even on Debian where it would be `dist-packages`). This breaks `cmake` packaging. There has been some discussion, as of yet unresolved. 

My preferred solution would be to package `eigenpy` directly as a CMake project with just a package.xml added. This _could_ be done if the `site-packages` hack gets removed. Or we could do it with a patch on top of the original project to set the Python lib path correctly.

v4hn, 2019-07-04:
Thank you for the swift response!

As you can probably see from my lament above, I'm also not happy with the current state of distribution of the library and it should definitely be improved!
However, I consider it rather unfortunate to drag the whole MoveIt project in there because of a single functional dependency.
Personally, I am not willing to spend work on that library given that `boost_numpy` is around the corner.

rhaschke, 2019-07-04:
@v4hn, what is your recommended approach then, given `boost::numpy` is available only since Bionic?
Should we implement a fall-back solution based on `boost::python` for < Bionic?

wxmerkt, 2019-07-04:
I wasn't aware it was pulled into such a large project in the first place. The whole cmake extras file is a hack around the original issue.
If you'd like to add weight behind the voice of `don't force override the Python install path on systems`, that would be greatly appreciated [the discussions so far didn't go far]. 

[I'm planning to re-release eigenpy, hpp-fcl, and pinocchio with the proper cmake packaging, even if necessary patched or forked to fix the python-package-path.]

v4hn, 2019-07-04:
@wxmerkt

> I'm planning to re-release eigenpy, hpp-fcl, and pinocchio with the proper cmake packaging, even if necessary patched or forked to fix the python-package-path.

It would definitely be appreciated over the current state.
A proper cmake config would help a lot to.

> don't force override the Python install path on systems

indeed.

@rhaschke 

> what is your recommended approach then, given boost::numpy is available only since Bionic?
Should we implement a fall-back solution based on boost::python for < Bionic?

I guess, a fall-back is a bad idea. As we already have problems having maintainers use and test older systems to begin with; we should probably avoid splitting code paths between system versions as much as possible.

If I had a recommended approach, I would probably file a pull-request over an issue. :)
Do you see any reasonable alternatives for now?
At least, I wanted to keep this issue open so we replace it by the boost library once we drop support for xenial in master. (I'm grateful to you you did not request this yet :+1:  )

rhaschke, 2019-07-04:
No, I don't have an easy solution as well.

rhaschke, 2019-07-06:
The issue #1524 might be related. @wxmerkt, could you please have a look to confirm this?

wxmerkt, 2019-07-06:
@rhaschke Is it possible that the eigenpy didnt correctly export the Numpy Include dirs? I am happy to re-release with a fork and the 'proper' packaging - but it will trigger downstream changes (switch to pkg-config). Would that be an option?

rhaschke, 2019-07-07:
I didn't look into the details of #1524, but either the numpy includes are in a different location when installed via pip, or they are missing at all / changed their name.
I'm happy to adapt to downstream changes. Just tell me, what has changed.

v4hn, 2019-07-08:
Yes, numpy is not included correctly, neither in upstream nor in the catkin version.

Also see OP:

> Instead upstream relies on pkg-config, but incidentally their pkgconfig file is broken too - it uses -isystem to include numpy which pkg-config --cflags-only-I eigenpy fails to extract.

According to my inquiries (which might be incomplete), the current configuration requires the symlink /usr/include/numpy to redirect to the "correct" headers. I had to add it on my system to get things to compile.

wxmerkt, 2019-07-11:
I am intending to re-release `eigenpy` with fixed `pkg-config` as a proper 3rd-party library. A prototype of this can be found in this PR: https://github.com/stack-of-tasks/eigenpy/pull/69. I have build locally (ROS Melodic) the latest MoveIt master and it is compiling/working fine - after applying a change to use `pkg_check_modules` (cf. #1550 for a demonstrator). What are your thoughts/comments/suggestions on this?