moveit/moveit #1540: "Failed to fetch current robot state" when running get_current_joint_values in python interface

adaofromnow, 2019-07-05:
### Description
i am a new Ros learner.
i am following  the "Move Group Python Interface"  tutorial .
i want make my swayer (UR )move  .
i have search on the internet and reading lots of answers ,but i still can't slove this problem(i meet this problem both in gazebo and real sawyer robot ):

when i run the code below
`  joint_goal = move_group.get_current_joint_values()`
i meet a error like this
> [ INFO] [1562331940.610865791, 6058.541000000]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1562331940.610894781, 6058.541000000]: Failed to fetch current robot state

i can follow this step-> " Planning to a Pose Goal"  to make my robot move,
but when i follow-> "Planning to a Joint Goal"  , i got error
my English is poor ,hope you can understand what i said . Thanks a lot for your help!!

### Your environment
* Kinetic
* Ubuntu 16.04.5 LTS
* Source

### Steps to reproduce
this is my code

>  moveit_commander.roscpp_initialize(sys.argv)
    rospy.init_node('move_group_python_interface_tutorial', anonymous=True)
    robot = moveit_commander.RobotCommander()
    scene = moveit_commander.PlanningSceneInterface()
    group_name = "right_arm"
    move_group = moveit_commander.MoveGroupCommander(group_name)
    display_trajectory_publisher = rospy.Publisher('/move_group/display_planned_path',
                                                              moveit_msgs.msg.DisplayTrajectory,queue_size=20)
    planning_frame = move_group.get_planning_frame()
    ~ ~~there is some print func i omit~~~~~~~
    print "============ Printing robot state"
    print robot.get_current_state()
    joint_goal = move_group.get_current_joint_values()

this is the move how i run my code

> $./intera.sh sim
   $roslaunch sawyer_gazebo sawyer_world.launch
   $rosrun intera_interface joint_trajectory_action_server.py 
   $roslaunch sawyer_moveit_config sawyer_moveit.launch electric_gripper:=true
   $rosrun intera_examples joint_torque_springs.py       //this is use toreset  arm  then  ctrl-c quit
   $rosrun intera_examples test.py      //run my code

### Expected behaviour
It should give me the information of the robot joint state 

### Actual behaviour
print ERROR 

### Backtrace or Console output

>  ros@ros-combol:~/rtest_ws$ rosrun intera_examples m.py 
[ INFO] [1562334484.381380421]: Loading robot model 'sawyer'...
[ WARN] [1562334484.381413912]: Skipping virtual joint 'world' because its child frame 'base' does not match the URDF frame 'world'
[ INFO] [1562334484.381425261]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1562334484.445300357, 8600.088000000]: Loading robot model 'sawyer'...
[ WARN] [1562334484.445323516, 8600.088000000]: Skipping virtual joint 'world' because its child frame 'base' does not match the URDF frame 'world'
[ INFO] [1562334484.445331716, 8600.088000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1562334485.357006257, 8601.000000000]: 
Deprecation warning: Trajectory execution service is deprecated (was replaced by an action).
Replace 'MoveGroupExecuteService' with 'MoveGroupExecuteTrajectoryAction' in move_group.launch
[ INFO] [1562334485.359397471, 8601.003000000]: Ready to take commands for planning group right_arm.
============ Planning frame: /world
============ End effector link: right_gripper_tip
============ Available Planning Groups: ['right_arm']
============ Printing robot state
[ WARN] [1562334486.366196966, 8602.009000000]: Joint values for monitored state are requested but the full state is not known
joint_state: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: "/world"
  name: [right_j0, head_pan, right_j1, right_j2, right_j3, right_j4, right_j5, right_j6, right_gripper_l_finger_joint,
  right_gripper_r_finger_joint]
  position: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  velocity: []
  effort: []
multi_dof_joint_state: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: "/world"
  joint_names: []
  transforms: []
  twist: []
  wrench: []
attached_collision_objects: []
is_diff: False
[ INFO] [1562334487.369318434, 8603.011000000]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1562334487.369348350, 8603.011000000]: Failed to fetch current robot state

rhaschke, 2019-07-06:
See same issue here: https://github.com/ros-planning/moveit/issues/1187#issuecomment-508891103

Hubert51, 2019-07-08:
Do you solve this problem?

adaofromnow, 2019-07-09:
thanks a lot for your answer .@Huber51 's answer in #1187  solved my problem

adaofromnow, 2019-07-09:
@rhaschke   @Hubert51   thanks a lot for your answer .
@Huber51 's answer in #1187 solved my problem.

I confused about @Hubert51's question in #1187  too.

closed 2019-07-09 by adaofromnow