moveit/moveit #1544: How to switch control mode between robot built-in mode and moveit control mode

Hubert51, 2019-07-08:
### Description
Currently, I can use moveit and baxter_python_interfacde to control the baxter. However, after I use moveit, I can not use baxter_python_interface to control baxter any more. The moveit tries to keep the current joint angle. So baxter_python_interface can not move the joint angle. If I use hand to control the baxter endeffector. The moveit will lose the control. Can you tell me how to let moveit lose control in python code. Thank you

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04

v4hn, 2019-07-08:
I'm sorry but we can't really help you with this, as it's specific to your robot setup.

> The moveit tries to keep the current joint angle

MoveIt does not do anything like that. The `move_group` node (which you probably use) sends action goals to the robot's `FollowJointTrajectory` action server. Once this action returns a result (you could check whether that's the case) MoveIt does not influence the robot anymore at all.

The controller on the robot decides how to coordinate its various interfaces, in your case whatever the `baxter_python_interface` uses and the mentioned action.

Maybe @IanTheEngineer wants to add a comment on the behavior of the Baxter though.

closed 2019-07-08 by v4hn

Hubert51, 2019-07-09:
Oh! I see. Thank you! Depend on your suggestion, I found the reason. It seems the baxter trajectory server tries to keep the joint angle from moveit. I write functions to start and stop trajectory node. Now it works pretty good for me. I can both use moveit and built-in control system to control the baxter