moveit/moveit #1546: eigenpy runtime dependency missing

jschleicher, 2019-07-09:
### Description
Moveit-commander is missing libeigenpy.so in kinetic binary release 0.9.16 (currently in ros-testing). I'd suggest to block the ongoing [Kinetic Sync](https://discourse.ros.org/t/preparing-kinetic-sync-2019-07-08/9818) and prepare a hotfix (either dropping b6de3e1d309df9f65d47b9ad0c2dc599a0f7b32c or adding the missing `run_depend`.

### Your environment
* ROS Distro: Kinetic
* Binary build 0.9.16

### Steps to reproduce
```
docker run -it ros:kinetic-ros-core

$ sed -i -r 's/ros\//ros-testing\//' /etc/apt/sources.list.d/ros-latest.list
$ apt update
$ apt install ros-kinetic-moveit-commander
$ python
>>> import moveit_commander
Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/__init__.py", line 4, in <module>
    from .move_group import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 41, in <module>
    from moveit_ros_planning_interface import _moveit_move_group_interface
ImportError: libeigenpy.so: cannot open shared object file: No such file or directory
>>> 

```

### Expected behaviour
Moveit-commander should be usable. See also PilzDE/pilz_industrial_motion#185.

gavanderhoorn, 2019-07-09:
I believe this is a duplicate of #1536.

jschleicher, 2019-07-09:
@gavanderhoorn I don't think so. #1536 is about getting rid of the dependency. This issue is about unusable binary package 0.9.16.

gavanderhoorn, 2019-07-09:
#1536 talks about the fact that the package is unusable on Kinetic. The only "solution" so far has been a proposal to remove the dependency completely.

That would seem to be what you are suggesting here as well.

jschleicher, 2019-07-09:
`apt install ros-kinetic-eigenpy` fixes the import issue for me, so the (preferred?) solution would be to add a `run_depend` to b6de3e1.

For me #1536 is mostly about packaging preferences...

gavanderhoorn, 2019-07-09:
I only noticed the `build_depend` ([here](https://github.com/ros-planning/moveit/blob/69a2b39853a500e2d3ffde7abc5c7d26f974d42e/moveit_ros/planning_interface/package.xml#L35)). 

There are more issues with `eigenpy` under Kinetic, but that `run_depend` should be added indeed.

(on `master` and `melodic-devel` a `depend` is used, so that captures both)

rhaschke, 2019-07-09:
Thanks for pointing out this issue. I will fix it and trigger a bugfix-release.

jschleicher, 2019-07-10:
Thank you @rhaschke for the fix!

closed 2019-07-10 by jschleicher