moveit/moveit #1547: Collision checking failing

pulkit123tak, 2019-07-09:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

pulkit123tak, 2019-07-09:
Hi 

I am using  ROS Kinetic with Ubuntu 16.04. I tried the moveit tutorials  given [here ](http://docs.ros.org/kinetic/api/moveit_tutorials/html/index.html). The franks_emika robot is colliding with the collision object. For the final pose, I put the end-effector in collision with the object and the robot trajectory was executed through RViz plugin. Here are the images:
 
[image]

[image]

Can someone why it is happening and how to avoid it ??

simonschmeisser, 2019-07-09:
It looks like this green box is just a marker? Markers are only for visualization not for collision checking. You need to add a scene object

pulkit123tak, 2019-07-09:
It is not a marker,it is a collision object. I can show the same without adding marker plugin. 
[image]
[image]

simonschmeisser, 2019-07-09:
Ok, sorry, my bad. Could you please edit your first post and fill in all of the questions asked in the template there?

rhaschke, 2019-07-09:
I wasn't able to replicate this issue with a fresh MoveIt Kinetic install (using the official MoveIt container)
and running the [MoveGroupInterface tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html#running-the-code). Please try to replicate your issue with a fresh install as well. Maybe you have some strange lib versions (e.g. of FCL) installed?

[image]

pulkit123tak, 2019-07-10:
Thanks @rhaschke , but I tried with installing ROS afresh and installing moveit and still the issue exsists

felixvd, 2019-10-11:
@pulkit123tak You might still have incompatible versions of other packages (e.g. FCL) installed. Have you tried running in the [Docker container](https://moveit.ros.org/install/docker/)?