moveit/moveit #1556: Control Cartesian Speed of End Effector

kwh5484, 2019-07-12:
### Description

There seems to be some interest in the ability to control the cartesian speed of an end effector (https://answers.ros.org/question/272459/how-to-control-cartesian-velocity/). I believe I have a valid way to do this and I would be happy to integrate it with the MoveIt source code if there is interest.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04

### Methodology
I didn't want to modify any existing MoveIt data structures or function calls, so I decided to modify the trajectory of the plan generated from a call to a _MoveGroupInterface_ object's _plan_ function before the plan is actually executed. Using the joint angles at each waypoint in the trajectory, I calculate the position of a specified end effector. I find the euclidean distance between the end effector in consecutive waypoints and divide this distance by a user provided speed to get the time difference between waypoints. I then use this time difference to update the waypoint _time_from_start_ attribute (provided that the new waypoint time does not violate joint velocity/acceleration constraints). Because of accelerations/decelerations, the user provided speed would be the end effector's average speed between waypoints. If anyone has any suggestions on ways to improve the algorithm, I'm definitely happy to hear them.

### Integration with MoveIt
If there is any interest in adding this to the MoveIt source code, I have thought of two ways to do so (although I am definitely open to suggestions). The first way is to leave it as it is now, a separate function that is called after _plan_ and before _execute_. 

Another way to do it would be to add end effector speed getters/setters in the _MoveGroupInterface_ code, and then modify the _plan_ function to change the waypoint timestamps after the trajectory is generated. I considered adding this functionality to moveit_core/trajectory_processing, but I'm not sure I possess the MoveIt knowledge to cleanly implement it this way.

v4hn, 2019-07-13:
Yes, there is still interest in this in general.

rhaschke, 2019-07-14:
I think the proper way of integrating such a feature into MoveIt is via a planning adapter for time parameterization. Following your simple algorithmic approach, one could run any of the joint-level algorithms first, which then would be followed by your adapter that will modify the timings.

Currently, the corresponding algorithms consider joint-level trajectories only, but they can be applied to Cartesian-level trajectories in the very same fashion (given there is only a single end-effector moving). 
Actually, your approach should also calculate/update accelerations and velocities. 
Also, remember that one usually needs acceleration/deceleration phases.

jschleicher, 2019-07-15:
Please also take a look at [PilzDE/pilz_industrial_motion](https://github.com/PilzDE/pilz_industrial_motion/tree/melodic-devel/pilz_trajectory_generation): It takes a list of (cartesian or joint) goals and moves with constant speed in between. It is implemented as a MoveIt capability. According to the linked question, this seems to be very similar to your use-case?

kwh5484, 2019-07-15:
@rhaschke thanks for the advice, I've added calculating/updating accelerations and velocities, and I'm working on adding acceleration/deceleration phases. 

@jschleicher I've looked over the package and I agree, it does seem to provide similar functionality. If the MoveIt community is happy with this implementation of controlling Cartesian speed, that's fine by me. After a quick read through of the README file, I can see two benefits of my implementation, but there may be more.
For one, my implementation allows for the max speed to be set explicitly (ex. I can set the end-effector to move at 0.2 m/s), whereas the pilz_industrial_motion package sets the end effector speed with a max velocity scaling factor. Also,my code doesn't replace the trajectory generator, it just modifies the existing trajectory. Again, this package may be fine for the community's needs, I just wanted to make my code available if there was interest in using it.

gavanderhoorn, 2019-07-15:
> For one, my implementation allows for the max speed to be set explicitly (ex. I can set the end-effector to move at 0.2 m/s), whereas the pilz_industrial_motion package sets the end effector speed with a max velocity scaling factor.

wouldn't `max_trans_vel * scaling_factor == effective_vel` and conversely `effective_vel / max_trans_vel == scaling_factor` (where `max_trans_vel` is a parameter of `PilzDE/pilz_industrial_motion`)?

Seems like @jschleicher and team had to make due with the structure of the `moveit_msgs::MotionPlanRequest` and used what was available to them.

kwh5484, 2019-07-15:
@gavanderhoorn that's true. If there's no interest, I can close the issue.

gavanderhoorn, 2019-07-15:
That was not the purpose of my comment.

I just wanted to make sure I understood yours.

I believe that what you're proposing certainly is interesting. Perhaps "seeing some code" would make for a more meaningful discussion?

v4hn, 2019-07-16:
The Pilz planner implements a number of nice algorithms that were not available before.
Still, algorithms in `moveit_core` that can modify trajectories in such ways are also very welcome.

Of course, if the same algorithms are implemented it would be nice to reuse code.

kwh5484, 2019-07-16:
I'd be happy to post my code. Unfortunately, it has to go through internal review before I can publicly release it, but that process is underway.

bumbastic, 2019-10-02:
I'm interested. any code comming?

kwh5484, 2019-10-03:
Sorry for the late response. I ended up going in another direction and not needing this feature entirely. Here's the code I was using. It's a little rough around the edges, but at the very least it might give you something to work off of.
```C++
void setAvgCartesianSpeed(moveit::planning_interface::MoveGroupInterface::Plan &plan, const std::string end_effector, const double speed)
{
    robot_model_loader::RobotModelLoader robot_model_loader("robot_description");
    robot_model::RobotModelPtr kinematic_model = robot_model_loader.getModel();
    robot_state::RobotStatePtr kinematic_state(new robot_state::RobotState(kinematic_model));
    kinematic_state->setToDefaultValues();
 
    int num_waypoints = plan.trajectory_.joint_trajectory.points.size();                                //gets the number of waypoints in the trajectory
    const std::vector<std::string> joint_names = plan.trajectory_.joint_trajectory.joint_names;    //gets the names of the joints being updated in the trajectory
 
    //set joint positions of zeroth waypoint
    kinematic_state->setVariablePositions(joint_names, plan.trajectory_.joint_trajectory.points.at(0).positions);
 
    Eigen::Affine3d current_end_effector_state = kinematic_state->getGlobalLinkTransform(end_effector);
    Eigen::Affine3d next_end_effector_state;
    double euclidean_distance, new_timestamp, old_timestamp, q1, q2, q3, dt1, dt2, v1, v2, a;
    trajectory_msgs::JointTrajectoryPoint *prev_waypoint, *curr_waypoint, *next_waypoint;
 
    for(int i = 0; i < num_waypoints - 1; i++)      //loop through all waypoints
    {
        curr_waypoint = &plan.trajectory_.joint_trajectory.points.at(i);
        next_waypoint = &plan.trajectory_.joint_trajectory.points.at(i+1);
         
        //set joints for next waypoint
        kinematic_state->setVariablePositions(joint_names, next_waypoint->positions);
 
        //do forward kinematics to get cartesian positions of end effector for next waypoint
        next_end_effector_state = kinematic_state->getGlobalLinkTransform(end_effector);
 
        //get euclidean distance between the two waypoints
        euclidean_distance = pow(pow(next_end_effector_state.translation()[0] - current_end_effector_state.translation()[0], 2) + 
                            pow(next_end_effector_state.translation()[1] - current_end_effector_state.translation()[1], 2) + 
                            pow(next_end_effector_state.translation()[2] - current_end_effector_state.translation()[2], 2), 0.5);
 
        new_timestamp = curr_waypoint->time_from_start.toSec() + (euclidean_distance / speed);      //start by printing out all 3 of these!
        old_timestamp = next_waypoint->time_from_start.toSec();
 
        //update next waypoint timestamp & joint velocities/accelerations if joint velocity/acceleration constraints allow
        if(new_timestamp > old_timestamp)
            next_waypoint->time_from_start.fromSec(new_timestamp);
        else
        {
            //ROS_WARN_NAMED("setAvgCartesianSpeed", "Average speed is too fast. Moving as fast as joint constraints allow.");
        }
         
        //update current_end_effector_state for next iteration
        current_end_effector_state = next_end_effector_state;
    }
     
    //now that timestamps are updated, update joint velocities/accelerations (used updateTrajectory from iterative_time_parameterization as a reference)
    for(int i = 0; i < num_waypoints; i++)
    {
        curr_waypoint = &plan.trajectory_.joint_trajectory.points.at(i);            //set current, previous & next waypoints
        if(i > 0)
            prev_waypoint = &plan.trajectory_.joint_trajectory.points.at(i-1);
        if(i < num_waypoints-1)
            next_waypoint = &plan.trajectory_.joint_trajectory.points.at(i+1);
 
        if(i == 0)          //update dt's based on waypoint (do this outside of loop to save time)
            dt1 = dt2 = next_waypoint->time_from_start.toSec() - curr_waypoint->time_from_start.toSec();
        else if(i < num_waypoints-1)
        {
            dt1 = curr_waypoint->time_from_start.toSec() - prev_waypoint->time_from_start.toSec();
            dt2 = next_waypoint->time_from_start.toSec() - curr_waypoint->time_from_start.toSec();
        }
        else
            dt1 = dt2 = curr_waypoint->time_from_start.toSec() - prev_waypoint->time_from_start.toSec();
 
        for(int j = 0; j < joint_names.size(); j++)     //loop through all joints in waypoint
        {
            if(i == 0)                      //first point
            {
                q1 = next_waypoint->positions.at(j);
                q2 = curr_waypoint->positions.at(j);
                q3 = q1;
            }
            else if(i < num_waypoints-1)    //middle points
            {
                q1 = prev_waypoint->positions.at(j);
                q2 = curr_waypoint->positions.at(j);
                q3 = next_waypoint->positions.at(j);
            }
            else                            //last point
            {
                q1 = prev_waypoint->positions.at(j);
                q2 = curr_waypoint->positions.at(j);
                q3 = q1;
            }
 
            if(dt1 == 0.0 || dt2 == 0.0)
                v1 = v2 = a = 0.0;
            else
            {
                v1 = (q2 - q1)/dt1;
                v2 = (q3 - q2)/dt2;
                a = 2.0*(v2 - v1)/(dt1+dt2);
            }
 
            //actually set the velocity and acceleration
            curr_waypoint->velocities.at(j) = (v1+v2)/2;
            curr_waypoint->accelerations.at(j) = a;
        }
    }
}

bumbastic, 2019-10-03:
Thanks, it builds and it works. Limiting the end effector speed to a constant speed.

kwh5484, 2019-10-03:
Great, glad to hear it.

felixvd, 2019-11-17:
I'd like to suggest implementing this feature into the main code base as a WMD2019 issue.

v4hn, 2019-11-18:
The proposed task is to implement a function in the `moveit_core/trajectory_processing` package that operates on a `moveit_msgs::RobotTrajectory` object to adjust the end-effector speed.

2scholz, 2019-11-20:
I am going to work on this.

onskateboard, 2021-01-13:
@keeganharris Thanks for your implementation. It works good, but be aware that you are limiting the speed to a desired speed factor. The speed of the endeffector isn't constant for waypoints where the acceleration limits of the individual joints leading to lower speeds then the desired one. To make it really constant you have to first  loop through all waypoints to find the minium possible speed for all waypoints (taking into account the acceleration constraints of each joint).

gautz, 2021-06-23:
I guess this could be closed in favor of #2674 .

v4hn, 2021-06-25:
> I guess this could be closed in favor of [...] .

This is an issue and it will be closed automatically upon merging the linked request

artemiialessandrini, 2021-09-06:
Actually when using `SesetAvgCartesianSpeed`, sometimes I get:
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
```
Could it be that update joint velocities/accelerations should go with
   ` for(int i = 1; i < num_waypoints; i++)`?

closed 2022-09-09 by rhaschke