moveit/moveit #1561: Collision sometime with the ground

Deastan, 2019-07-15:
Hello everyone,

My environment

    ROS Distro: [Indigo]
    OS Version: e.g. Ubuntu 14.04
    Source or Binary build?
    If binary, which release version? install with : sudo apt-get install ros-indigo-moveit => ros-indigo-moveit is already the newest version.
Gazebo 7 (before Gazebo 2.2 and also the same issue)

Do someone has an idea why the robot collides with the ground using moveit ? I am doing reinforcement learning and everything is working well except that part. I can go everywhere I need to go in cartesian space or using joint space. The end is working well too.
But sometime when I work close to the ground and go from (x = 0.5, y = 0.1, z = 0.1) to (x = 0.5, y = 0.2, z = 0.1), it touch the ground. The robot then (a kuka iiwa lbr 7) deformed itself with the constraint and finally reach the point in more than 5 to 10 sec. I know it can reach those points with others paths but I don't understand why it take this one.
I am working (unfortunately) on Ubuntu 14, with ROS indigo. I am using as solver: RRTkConfigDefault or maybe a bit better RRTConnectkConfigDefault.

[image]

I can give you more information if you need it.

Kind regards,

Jonathan

gavanderhoorn, 2019-07-15:
Do you model your environment in your urdf / does MoveIt 'know' there is a ground plane there?

Deastan, 2019-07-15:
Seriously ? Is it so simple and me so s...pid ? :D
I should do it by adding in the Step 3: Add Virtual Joints of the setup assistant? Or how can I do in the URDF file ?

gavanderhoorn, 2019-07-15:
How do you expect the planners to 'know' about any obstacles if the robot is simply floating in empty space in the description that you give them?

> Or how can I do in the URDF file ?

Add a flat `box` under the robot?

gavanderhoorn, 2019-07-15:
I'm going to close this as it's likely to be a user-side configuration issue.

If it's not, we can re-open.

Feel free to keep commenting on the issue of course.

closed 2019-07-15 by gavanderhoorn

Deastan, 2019-07-15:
Thank you very much for you answer! I am really sorry for this mistake.

gavanderhoorn, 2019-07-15:
>  I am really sorry for this kind of mistake.

no need to apologise.