moveit/moveit #1565: No collision checks within its own world

j-petit, 2019-07-16:
As I wrote a [test-case](https://github.com/j-petit/moveit/blob/444a7bfd4e286482882a09eadaf15af939e80738/moveit_core/collision_detection/include/moveit/collision_detection/test_collision_common_panda.h#L152) for checking collision within a single `CollisionWorld` environment, I noticed that there does not exist the possibility to check collisions within the `CollisionWorld` itself - only to [other `CollisionWorld`s](https://github.com/ros-planning/moveit/blob/56b7c61a88287bf027415d96ce038de668d2e05a/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_world_fcl.h#L70) (using the FCL collision checking implementation).

Personally, I always assumed this is a very basic functionality and should be included, but maybe I am missing something why this is not necessary (or maybe it is implemented and I just don't see where?).

BryceStevenWilley, 2019-07-24:
Sorry, I missed this issue. If you have a some spare time while working on documentation you could add that functionality. I don't see any problems with the method, but I also don't quite see the benefits. Often, items in the world will be defined very closely together and sometimes overlapping, so I don't think that colliding objects in a single world print a warning or something like that. If it's something that might be useful for user defining collision worlds, just as an available query, just having it in C++ might not be as useful, and it might need to go in the python interface as well.

j-petit, 2019-07-24:
You're right does not make too much sense. Closing this.

closed 2019-07-24 by j-petit