moveit/moveit #1568: /compute_ik fails but move_group move function suceeds!

gstavrinos, 2019-07-16:
### Description

I am developing a package that allows for an easier pick and place procedure. The package used to work without problems, but now it appears that the IK tests always fail.
More specifically, when I call the /compute_ik service to test if the ee position is valid, before requesting a plan, I never get a solution. The weird thing is that if I command the robot to plan to that seemingly invalid position, it succeeds!

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Latest Source Build

### Steps to reproduce
This is a mwe of the problem:
```
#!/usr/bin/env python
import sys
import rospy
import moveit_commander
from geometry_msgs.msg import PoseStamped
from moveit_msgs.srv import GetPositionIKRequest, GetPositionIK

def main():
    moveit_commander.roscpp_initialize(sys.argv)
    rospy.init_node("mwe")

    compute_ik_srv = rospy.ServiceProxy("/compute_ik", GetPositionIK)
    rospy.wait_for_service("/compute_ik")

    arm_move_group = moveit_commander.MoveGroupCommander("arm")
    robot_commander = moveit_commander.RobotCommander()

    pose = PoseStamped()
    pose.header.frame_id = "world"
    pose.header.stamp = rospy.Time.now()
    pose.pose.position.x = -0.285039418823
    pose.pose.position.y = -0.128284967952
    pose.pose.position.z = 1.10245513838
    pose.pose.orientation.x = 0.287262476532
    pose.pose.orientation.y = 0.290124446559
    pose.pose.orientation.z = -0.644487562197
    pose.pose.orientation.w = 0.646470940769

    req = GetPositionIKRequest()
    req.ik_request.group_name = "arm"
    req.ik_request.robot_state = robot_commander.get_current_state()
    req.ik_request.avoid_collisions = True
    #req.ik_request.ik_link_name = arm_move_group.get_end_effector_link()
    req.ik_request.pose_stamped = pose
    k = compute_ik_srv(req)
    print k.error_code.val == 1

    arm_move_group.set_pose_target(pose)
    print arm_move_group.go()

main()

```

### Expected behaviour
Both prints should either be the same value, or at least, when the IK result is False, the planning result should be False too.

### Actual behaviour
The first print is False but the second print is True, which in my mind should not be possible, since the planning procedure includes the IK calculations.

### Extras
This is the pick and place package repo:

https://github.com/gstavrinos/ez_pick_and_place

and this is our robot repo:

https://github.com/gstavrinos/ez_pick_and_place

gstavrinos, 2019-07-18:
Just some extra information:

I tested the same code (with some variations on goal points and move group names) with various manipulators, like the UR5 and the Ffanuc m16, with exactly the same results.

gstavrinos, 2019-07-22:
Anyone looking into this? If there is something more I can provide you with, to help you understand my problem, I would be more than glad to do so!

Thanks!

ptriantd-ocado, 2020-01-12:
Hi, @gstavrinos I am having the same issue. Did you eventually figure out why it is happening?

gstavrinos, 2020-01-13:
Hey @ptriantd-ocado, unfortunately no, but as I was advised, check your robot's description for changes that might have occurred since the last time it worked for you.

Please keep me update if you find any solution to this.

ConradLiuuu, 2021-08-17:
Hi gstavrinos, I was struggling in this issue too.
Finally I found the problem in poseStamped.header.frame_id
In my situation the "Link0" is base_link as shown in attached image, so the frame_id should be specified base_link instead of world.

Hope this suggestion can help you.

[image]

gstavrinos, 2021-08-18:
Thanks for your input @ConradLiuuu . As stated in my previous comment, it appears that some differences in the robot description caused the failure of the `/compute_ik` request. Still, this shouldn't be happening, thus I am leaving this issue open.