moveit/moveit #1570: Dropping MoveIt 'master' branch support of ROS Kinetic

davetcoleman, 2019-07-18:
@rhaschke commented in https://github.com/ros-planning/moveit/pull/1504#issuecomment-512268617

> Looks like Kinetic/Xenial ships with an older Bullet version. I suggest dropping Kinetic support for now.

In reference to @j-petit and @ommmid 's efforts to add Bullet and TrajOpt to MoveIt. 

I'm moving the discussion here on whether this is a good idea, and I would like to finalize this discussion at our next [Maintainer Meeting](https://discourse.ros.org/t/moveit-maintainer-meeting-all-invited-july-25th/9899/3).

From the top of my head:

*Pros*
- Avoid a lot of extra backwards compatible support and speed up MoveIt progress
- Allow access to latest features of dependencies like Bullet
- ROS Kinetic was released in 2016 and is old 

*Cons*
- Many companies are still on Ubuntu 16.04
- Some of PickNik's clients are using 'master' + Ubuntu 16.04
- Less developers able to contribute to 'master' branch

Are there other options? I believe Tesseract maintained Ubuntu 16.04 support by building Bullet from source, right? 

What do you think?

j-petit, 2019-07-18:
> Are there other options? I believe Tesseract maintained Ubuntu 16.04 support by building Bullet from source, right?

Yes, tesseract did a build from source.

rhaschke, 2019-07-18:
I wasn't proposing to drop Kinetic support in general, but only for this feature-branch (_for now_).
My long-term goal (before merging the feature branch into master) would be to check the available Bullet version and decide whether to compile bullet-collision or not. As this is an (optional) plugin, we don't need to support it on Xenial, but the rest of MoveIt we should.
Same argumentation holds for the TrajOpt planner plugin.

felixvd, 2019-07-21:
Agreed. The check for the available Bullet version seems very reasonable to me, and could be implemented before the merge of the feature branch. 

Also, since the immediate performance benefit of the changes are unclear and the main benefits appear to be in the future. I would focus on developing the feature in Melodic and ignore Kinetic support in the feature branch for now.

closed 2019-07-22 by rhaschke

j-petit, 2019-11-16:
I'd like to open this issue up again as all PRs to the feature branch are merged and we can discuss merging the feature branch into master.

davetcoleman, 2019-11-20:
I don't think this issue is needed to be re-opened, because we don't need to drop support of 16.04, after all, right? Instead your [issue](https://github.com/ros-planning/moveit/pull/1744) just disabled Bullet on Kinetic