moveit/moveit #1576: RuntimeError: Unable to connect to move_group action server 'move_group' wit

zouguonideshiguang, 2019-07-21:
### Description

PluginManager._load_plugin() could not load plugin "rqt_easy_handeye/Hand-eye Calibration automatic movement":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 101, in load
    return class_ref(plugin_context)
  File "/home/he/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 296, in __init__
    self._widget = CalibrationMovementsGUI()
  File "/home/he/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 149, in __init__
    self.local_mover = CalibrationMovements(move_group_name, max_velocity_scaling, max_acceleration_scaling)
  File "/home/he/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibrationmovements.py", line 26, in __init__
    self.mgc = MoveGroupCommander(move_group_name)
  File "/home/he/catkin_ws/src/moveit_commander/src/moveit_commander/move_group.py", line 51, in __init__
    self._g = _moveit_move_group_interface.MoveGroup(name, robot_description)

RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)

### Your environment
* ROS Distro:|Kinetic]
* OS Version:  Ubuntu 16.04
* Source or Binary build?
*

closed 2019-07-21 by zouguonideshiguang

RDaneelOlivav, 2019-12-10:
Is this issue solved in some way? Does someone know how to change this allotted time?
How can we get more info on the reason why this is not connecting?

akgoins, 2020-01-08:
The error comes from the `MoveGroupInterface::waitForAction` function [called here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L149).  If a string is passed into the constructor of the `MoveGroupInterface`, then the default timeout value is infinite.  If another timeout value is desired, use one of the other constructors that takes the [MoveGroupInterface::Options](http://docs.ros.org/kinetic/api/moveit_ros_planning_interface/html/structmoveit_1_1planning__interface_1_1MoveGroupInterface_1_1Options.html) struct.  I came across this error myself because I was placing my node in a namespace in the launch file.  When trying to connect to the action servers, the node handle used was resolving the action name to the relative namespace, thus my node was looking for the `/my_node/move_group` action when the action was actually just `/move_group`.  The fix for me was to remove the namespace in my node or to remap all of the actions for my node to the global namespace (e.g. `<remap from="/my_node/move_group" to="/move_group" />`).  I also tried passing in `ros::NodeHandle("/")` as part of the interface options, but the global node handle did not resolve the name spacing issue.  Maybe someone else has a better solution to this.

makangzhe, 2020-10-22:
@zouguonideshiguang  do you fix it ?  can you tell me how to resolve the problem?

QuantuMope, 2021-04-11:
More of a workaround than a fix, but you can get this to work reliably by delaying the calibrate launch file by using `timed_roslaunch`. Below is a sample.

    <include file="$(find timed_roslaunch)/launch/timed_roslaunch.launch">
        <arg name="time" value="5" />
        <arg name="pkg" value="easy_handeye" />
        <arg name="file" value="calibrate.launch" />
        <arg name="value" value="eye_on_hand:=false move_group:=right_arm
                                 tracking_base_frame:=cam_1_link tracking_marker_frame:=marker
                                 robot_base_frame:=base robot_effector_frame:=right_gripper_base">

yananliusdu, 2023-04-14:
same problems here not solved yet withou abovementioned solutions