moveit/moveit #1577: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION

lusu8892, 2019-07-22:
### Description

moveit_core can not built, because of EIGEN_STATIC_ASSERT error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source build
* commit: 43076bd, kinetic-devel

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
moveit_core should be catkin built

### Actual behaviour
moveit_core can not be build

### Backtrace or Console output
`Errors     << moveit_core:make /home/xxxx/ws_moveit/logs/moveit_core/build.make.009.log             
In file included from /usr/include/eigen3/Eigen/Geometry:44:0,
                 from /home/xxxx/ws_moveit/src/moveit/moveit_core/distance_field/include/moveit/distance_field/distance_field.h:48,
                 from /home/xxxx/ws_moveit/src/moveit/moveit_core/distance_field/src/distance_field.cpp:37:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/xxxx/ws_moveit/src/moveit/moveit_core/distance_field/src/distance_field.cpp:212:23:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
make[2]: *** [distance_field/CMakeFiles/moveit_distance_field.dir/src/distance_field.cpp.o] Error 1
make[1]: *** [distance_field/CMakeFiles/moveit_distance_field.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/include/eigen3/Eigen/Geometry:44:0,
                 from /home/xxxx/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:47,
                 from /home/xxxx/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h:41,
                 from /home/xxxx/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:47,
                 from /home/xxxx/ws_moveit/src/moveit/moveit_core/kinematic_constraints/include/moveit/kinematic_constraints/kinematic_constraint.h:40,
                 from /home/xxxx/ws_moveit/src/moveit/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp:37:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/xxxx/ws_moveit/src/moveit/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp:323:74:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
make[2]: *** [kinematic_constraints/CMakeFiles/moveit_kinematic_constraints.dir/src/kinematic_constraint.cpp.o] Error 1
make[1]: *** [kinematic_constraints/CMakeFiles/moveit_kinematic_constraints.dir/all] Error 2
make: *** [all] Error 2
`

rhaschke, 2019-07-22:
Please follow the [instructions to install from source](https://moveit.ros.org/install/source/#download-source-code). The master branch switched from Eigen::Affine to Eigen::Isometry matrices and thus requires several other packages to be built from source as well.

lusu8892, 2019-07-22:
I followed this instruction, but some functions from robot_state library is no longer available to be used, such as computeCartesianPath, setFromIK(some function overloaded no longer exists)

rhaschke, 2019-07-22:
There are many API changes since Kinetic. Please have a look into the [migration notes](https://github.com/ros-planning/moveit/blob/master/MIGRATION.md)

lusu8892, 2019-07-22:
Thanks for your reply!
However, I want to keep using Kinetic version of MoveIt! which has some important function such as computeCartesianPath in robot_state header. However, the master branch of moveit seems no longer support that.
So, when you said all required upstream packages needs to be built, which branch of those packages should I clone in local?

rhaschke, 2019-07-22:
The package versions are stated in moveit.rosinstall. Please follow the instructions given in the migration notes. The master branch still builds on Kinetic. If you rely on deprecated or removed functions, you should keep the kinetic-devel branch of MoveIt.

BryceStevenWilley, 2019-07-24:
@lusu8892 Note that [computeCartesianPath](https://github.com/ros-planning/moveit/blob/ff50476c4070eb86d0a70aa39281d5805db13fa5/moveit_core/robot_state/src/cartesian_interpolator.cpp#L53) still exists, it's just been refactored into a separate cartesian interpolater class. [setFromIK](https://github.com/ros-planning/moveit/blob/471ebbe7918786f8d6f6ba69cce1f0001ddb2a2c/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L916) also still exists, just the `attempts` parameter was deprecated.