moveit/moveit #1581: PlanningScene: "Move for attached objects not yet implemented"

francofusco, 2019-07-24:
### Description

This issue is just to report a behavior that in my opinion could be handled differently. When an object is currently attached to a link, it cannot be moved using the "intended workflow". What I mean is that when an object is processed by `PlanningScene::processAttachedCollisionObjectMsg(const moveit_msgs::AttachedCollisionObject &object)`, if `object.object.operation` is `MOVE` we get the error `Move for attached objects not yet implemented`. However, note that if the operation is `ADD`, this effectively causes the pose to be updated.

### Environment

* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary build
* If binary, which release version?

### Steps to reproduce

The [Planning Scene ROS API](https://ros-planning.github.io/moveit_tutorials/doc/planning_scene_ros_api/planning_scene_ros_api_tutorial.html) tutorial can be used as an example. After attaching the object to the robot, the following code can be added:
```c++
for(int i=0; i<10 && ros::ok(); i++) {
  pose.position.z += 0.05;
  ROS_INFO("Changing position to [ %f %f %f ]", pose.position.x, pose.position.y, pose.position.z);
  attached_object.object.primitive_poses[0] = pose;
  attached_object.object.operation = attached_object.object.MOVE; // <-- ADD makes this work
  planning_scene.robot_state.attached_collision_objects[0] = attached_object;
  planning_scene.robot_state.is_diff = true;
  planning_scene_diff_publisher.publish(planning_scene);
  ros::Duration(0.5).sleep();
}

  visual_tools.prompt("Press 'next' in the RvizVisualToolsGui window to continue the demo");
```

### Proposed Workaround

Moving the attached object is actually feasible when using the `ADD` operation. I suggest to remove from `PlanningScene::processAttachedCollisionObjectMsg` [these lines](https://github.com/ros-planning/moveit/blob/895ded362423a4e4c01eeece30baf30cacef15d7/moveit_core/planning_scene/src/planning_scene.cpp#L1680):
```c++
else if (object.object.operation == moveit_msgs::CollisionObject::MOVE)
{
  ROS_ERROR_NAMED(LOGNAME, "Move for attached objects not yet implemented");
}
```
and to let the update happen as if the operation was `ADD`, perhaps by printing a warning on the console that explains that a dedicated move functionality is not implemented, but still allowing the pose to be updated.

If you feel this is worth a PR, I can make one during next weeks when I have more time!

felixvd, 2019-07-26:
As you can only attach objects to robot links and not to other objects, for the moment it really should do the same thing as ADD. I don't think we will add child objects anytime soon. A PR would be welcome, I will make sure to review it.

felixvd, 2019-10-11:
This is a good first issue for someone. Can someone tag it for WMD2019? @davetcoleman

v4hn, 2019-10-16:
@felixvd please do that yourself, see the two labels I added.

JafarAbdi, 2019-11-20:
I am going to work on this.

davetcoleman, 2019-11-20:
Only maintainers (github write access) have ability to add labels. I would still like to add more maintainers to our list, including @felixvd

grimmmyshini, 2020-03-03:
I would like to put up a PR for this if the issue is still open/relevant

felixvd, 2020-03-03:
It is, but take note of #1866 and #1867 if you work on this.

sherinee, 2020-03-13:
I have the same problem while using the Move Group Python Interface. I want to be able to change the orientation of a collision object (which was added previously). I try to assign a new pose to the object but there is no change in the pose of the object. Any help would be greatly appreciated!

v4hn, 2020-03-13:
You can always do that using `ADD` over `MOVE` for the operation.
The main difference is that the former will parse and update all collision shapes (usually one mesh or primitive) for every operation.

Everyone is very welcome to write a pull-request that updates the poses of existing attached objects of the current `RobotState` on `MOVE`.
The relevant line is obviously here:
https://github.com/ros-planning/moveit/blob/master/moveit_core/planning_scene/src/planning_scene.cpp#L1652