moveit/moveit #1585: move_group process dying during load

cpetersmeier, 2019-07-28:
### Description
After a fresh install of Ubuntu, ROS and MoveIt I try to plan motions for the Fanuc M20IA Robot using the ROS-Industrial Repository. 
When I launch the moveit_planning_execution.launch file, the move_group node immediately shuts down. Therefore, the other nodes that depend on it do not function correctly.
I think I tracked it down to be a MoveIt issue, since I also get errors when I try to start the official tutorials.
I installed Ubuntu in Dualboot mode beside Windows. Ubuntu has its own harddrive in the system with approx. 450 GB of space.

### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04.2 LTS
* Binary build / Source build
* MoveIt Version: 1.0.1-0bionic.20190628.142221 500

### Steps to reproduce
1. Install fresh Ubuntu copy
2. Install ROS and MoveIt from the official tutorial pages.
3. Install sudo apt-get ros-melodic-moveit*
4. Create a new workspace, clone ros-i fanuc [repository](https://github.com/ros-industrial/fanuc)
5. execute ` roslaunch fanuc_m20ia_moveit_config moveit_planning_execution.launch sim:=true debug:=true`

### Expected behaviour
Startup of RVIZ, Startup of move_group node, load of the robot URDF into the parameter server.
System should be ready for planning with the RVIZ GUI, the MoveIt Commander or the API.

### Actual behaviour
During the execution of the launch file, the move_group node immediately dies.
The robot description gets correctly loaded and displayed.
I am able to drag the robot around in RVIZ, using the interactive markers.
I am not able to choose planners (field is empty)

### Backtrace or Console output
The console output can be found [here](https://gist.github.com/cpetersmeier/cc55b64afeb4341dd737f42af82cf30b)

felixvd, 2019-07-29:
1) `roslaunch fanuc_m20ia_moveit_config moveit_planning_execution.launch sim:=true` expects a simulator like Gazebo to be running, as far as I know. Have you tried `roslaunch fanuc_m20ia_moveit_config demo.launch`?

2) If that does not work, have you tried running the tutorials in a [docker container](https://moveit.ros.org/install/docker/) to confirm that it's not your system installation that is the issue?

cpetersmeier, 2019-07-29:
1. As far as I am aware, the `moveit_planning_execution.launch` file does not use gazebo, rather it uses only rviz. I installed Ubuntu on a virtual machine using vm-ware and windows in dualboot mode as a host system on the same pc. Everything runs.

2. As I have not used docker before, I am a little unsure how to proceed: I started the MoveIt container with `docker run -it moveit/moveit:melodic-release`, initialized a workspace in the /home directory, `git clone` the ROS-I Fanuc repository, but when I follow its installation instructions [here](https://github.com/ros-industrial/fanuc/blob/indigo-devel/readme.md) cannot install the dependencies.

I observed, that after a system reboot, the `fanuc_m20ia_moveit_config demo.launch` works for one time,throwing an error due to a missing library. After that, the original error appears every time.

cpetersmeier, 2019-07-29:
In the `move_group.launch` file from the _fanuc_m20ia_moveit_config_ folder from the ros industrial fanuc repository, I set the GDB debug option to true.
I pasted the output [here](https://gist.github.com/cpetersmeier/481d5f785c5ce063c5f09fe5729dd7d8).
I could see the issue comming from this error and having something todo with the EigenMatrix Class:
`tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device`
I followed the instuctions in the second answer of [this](https://answers.ros.org/question/48056/using-roslaunch-with-gdb-gives-error-tcsetpgrp-failed-in-terminal_inferior-inappropriate-ioctl-for-device/) threat and post [here](https://gist.github.com/cpetersmeier/e248e9c11c000afba8cbe4411cb24c36) a detailed error description from xterm with bg.

felixvd, 2019-07-30:
> As far as I am aware, the `moveit_planning_execution.launch` file does not use gazebo, rather it uses only rviz.

The `moveit_planning_execution.launch` file does not start Gazebo, but it connects to external controllers, so it would expect either real robots or robots simulated with Gazebo (most commonly) connected to the system. The `demo.launch` file includes fake controllers so you can move your robot and do planning out of the box.

> I started the MoveIt container with `docker run -it moveit/moveit:melodic-release`, initialized a workspace in the /home directory...

The container comes with a workspace built and initialized in `/root/ws_moveit`. Just clone your package into there, install the dependencies and build it again.

> when I follow its installation instructions here cannot install the dependencies.

What keeps you from installing them? Is there an error? Try executing `apt-get update && rosdep install --from-paths src --ignore-src -r -y` in the `ws_moveit` directory to install all the dependencies of your packages.

Please also try `roslaunch panda_moveit_config demo.launch` in the docker, to confirm that things are generally working. I can't really tell what the other errors mean yet.

cpetersmeier, 2019-07-31:
After a reinstall of Ubuntu and ROS / MoveIt, from the MoveIt Docker Package I started this gui enabled docker:

`sudo ./gui-docker -it --rm moveit/moveit:melodic-source /bin/bash`

I tried `roslaunch panda_moveit_config demo.launch`. Unfortunately it also crashed. I made a Gist with the Console output (here)[https://gist.github.com/cpetersmeier/83ab1b85a7407b62e56c51c387ba57fd].

felixvd, 2019-08-01:
Try `apt-get update && rosdep install --from-paths src --ignore-src -r -y`, then `catkin build` and `roslaunch panda_moveit_config demo.launch`. That should already be done in the docker container (which I see you're in), but still.

I am confused by [these lines](https://gist.github.com/cpetersmeier/83ab1b85a7407b62e56c51c387ba57fd#file-gistfile1-txt-L219) in your output. They did not appear in your original gist. Maybe they will be clearer after the other package errors are fixed, or if you set the logger level to DEBUG.

cpetersmeier, 2019-08-01:
I ran the container again with
`sudo ./gui-docker -it --rm moveit/moveit:melodic-source /bin/bash`
then I installed the fanuc ros i packages again like so:
`git clone -b indigo-devel https://github.com/ros-industrial/fanuc.git src/fanuc`
`rosdep update`
`apt-get update && rosdep install --from-paths src --ignore-src -r -y`
`catkin build`
I then ran 
`roslaunch panda_moveit_config demo.launch` . It did not have the visualTools installed,
but move_group did not crash! Here is the output:
[Output](https://gist.github.com/cpetersmeier/28a21bbd4c0c7b8d351bb23c4fe641e4) **before** sourcing devel/seup.bash

I then sourced the devel/setup.bash to update my workspace after the install and to run fanuc packages. I tried the panda demo one more time. move_group crashed.

[Output](https://gist.github.com/cpetersmeier/8830880862c861d9deb69900809c997a) **after** sourcing devel/setup.bash

I am wondering if I am cloning a non workable version of the fanuc packages, which then destroy
my system somehow.

gavanderhoorn, 2019-08-01:
> The `moveit_planning_execution.launch` file does not start Gazebo, but it connects to external controllers, so it would expect either real robots or robots simulated with Gazebo (most commonly) [...]

just an FYI: `sim:=true` makes `moveit_planning_execution.launch` use the `industrial_robot_simulator`. It does not depend on Gazebo or anything else.

`sim:=false` would set things up such that an external `FollowJointTrajectory` action server needs to be running.

gavanderhoorn, 2019-08-01:
```
[ INFO] [1564680395.989878878]: Using planning request adapter '@???]
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

this would seem to be the cause (or at least suspicious).

There is no `'@???]` planning request adapter. That string looks like random memory or an uninitialised pointer is being printed.

> I am wondering if I am cloning a non workable version of the fanuc packages, which then destroy
> my system somehow.

I regularly use packages from `ros-industrial/fanuc` on a Melodic system and do not encounter the described issue. That is `n==1` as far as 'evidence', but I'm not aware of anything in those packages that could cause this.

You may want to take a look at the output of the `rosdep install` step you run. It may be installing 'strange' or unexpected packages.

cpetersmeier, 2019-08-01:
I copied the output of the `rosdep install` command for reference [here](https://gist.github.com/cpetersmeier/00017f4aac823a214f390fae9c9ed0aa) 
I used:
` apt-get update && rosdep install --from-paths src --ignore-src -r -y` install.

gavanderhoorn, 2019-08-01:
That all looks normal and expected.