moveit/moveit #1596: stack smashing

IndustrialEngStudent, 2019-08-02:
### Description
I tried to install the IKFast solver for my ur5 robot. Unfortunately, I gave up on this due to the many errors and the time limit of my project. The problem now is that ever since I did these installations, my old .launch & .py file stopped working due to 'stack smashing´. For instance when I am trying:
`$ roslaunch moveit_setup_assistant setup_assistant.launch `

How can I fix this?
### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Binary build

### Backtrace or Console output

```
$ roslaunch moveit_setup_assistant setup_assistant.launch
... logging to /home/laurah/.ros/log/3d8fb5f2-b4fe-11e9-af96-288023d628d5/roslaunch-laurah-HP-EliteBook-840-G1-25257.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laurah-HP-EliteBook-840-G1:33291/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

ROS_MASTER_URI=http://localhost:11311

process[moveit_setup_assistant-1]: started with pid [25277]
[rospack] Error: no package given
[librospack]: error while executing command
[ INFO] [1564735905.209161210]: Loaded ur5_robotiq_2f85 robot model.
[ INFO] [1564735905.209293765]: Setting Param Server with Robot Description
[ERROR] [1564735905.209763044]: [setParam] Failed to contact master at [localhost:11311].  Retrying...
[ INFO] [1564735905.267255612]: Connected to master at [localhost:11311]
[ INFO] [1564735905.272991043]: Robot semantic model successfully loaded.
[ INFO] [1564735905.273043406]: Setting Param Server with Robot Semantic Description
[ INFO] [1564735905.318224859]: Loading robot model 'ur5_robotiq_2f85'...
[ INFO] [1564735905.318291040]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1564735906.049074072]: Stereo is NOT SUPPORTED
[ INFO] [1564735906.049178696]: OpenGl version: 3 (GLSL 1.3).
[ INFO] [1564735906.129111040]: Loading Setup Assistant Complete
[ INFO] [1564735906.203534522]: Loading robot model 'ur5_robotiq_2f85'...
[ INFO] [1564735906.203580514]: No root/virtual joint specified in SRDF. Assuming fixed joint
*** stack smashing detected ***: /opt/ros/kinetic/lib/moveit_setup_assistant/moveit_setup_assistant terminated
================================================================================REQUIRED process [moveit_setup_assistant-1] has died!
process has died [pid 25277, exit code -6, cmd /opt/ros/kinetic/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/home/laurah/.ros/log/3d8fb5f2-b4fe-11e9-af96-288023d628d5/moveit_setup_assistant-1.log].
log file: /home/laurah/.ros/log/3d8fb5f2-b4fe-11e9-af96-288023d628d5/moveit_setup_assistant-1*.log
Initiating shutdown!
```
================================================================================
[moveit_setup_assistant-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

v4hn, 2019-08-29:
Sorry for the late feedback, I guess you already resolved this one way or another.

As you might imagine, this behavior does not make any sense from our perspective and you do not have clear pre-conditions to trigger this.
I hope reinstalling helped?

A gdb backtrace might have been helpful here.

Closing. Feel free to reopen if this is still relevant.

shyamashi, 2019-11-15:
I am experiencing the exact same issue. But I could not find a solution. 

- ROS Kinetic, Ubuntu 16.04

Steps to reproduce:
1. My ``` urdf, urdf.xacro ``` files  available at``` https://github.com/shyamashi/free_floating_3_dof_arm/tree/master/urdf  ```
2. ``` roslaunch moveit_setup_assistant setup_assistant.launch ```
3. Loaded the above ``` urdf.xacro ``` file. MoveIt shows the robot and when I go to ``` Self-Collisions--> Generate Collision Matrix ```, it crashes and leaves the exact same error as above
4. I tried with ``` panda_arm_hand.urdf.xacro ``` available at ``` https://github.com/frankaemika/franka_ros/tree/kinetic-devel/franka_description/robots ``` and it is able to generate collision matrix but it threw the same error when I tried with ```ur5_robot.urdf.xacro```

Any help would be really appreciated

v4hn, 2019-11-15:
@shyamashi Thanks for providing instructions to reproduce, they will hopefully help someone to investigate the problem soon.

rhaschke, 2019-11-17:
The issue is not reproducible on MoveIt master.

rhaschke, 2019-11-17:
I cannot reproduce the issue on kinetic-devel either. @shyamashi, is this issue deterministic or random for you? If you can reproduce the error, please provide a stack trace from a debugger. Thanks.

rhaschke, 2019-11-17:
Also using an address sanitizer, I wasn't able to trigger the problem.

shyamashi, 2019-11-18:
Thanks @rhaschke for looking into it. 
I built MoveIt from source as instructed in [here](https://moveit.ros.org/install/source/). A video of what exactly is happening is [here](https://github.com/shyamashi/free_floating_3_dof_arm/blob/master/stack_smashing.mp4). The terminal output is 
```
 roslaunch moveit_setup_assistant setup_assistant.launch 
... logging to /home/ar0058/.ros/log/f893d60e-09f3-11ea-89ab-6c2b59d30063/roslaunch-clws00083-54433.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://clws00083:45432/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

auto-starting new master
process[master]: started with pid [54444]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f893d60e-09f3-11ea-89ab-6c2b59d30063
process[rosout-1]: started with pid [54457]
started core service [/rosout]
process[moveit_setup_assistant-2]: started with pid [54460]
[ERROR] [1574075409.424188926]: poll failed with error Interrupted system call
[ERROR] [1574075409.424629843]: poll failed with error Interrupted system call
[ERROR] [1574075409.427273303]: poll failed with error Interrupted system call
[ERROR] [1574075409.427651261]: poll failed with error Interrupted system call
[rospack] Error: no package given
[librospack]: error while executing command
[ INFO] [1574075420.948712996]: Loaded free_floating_spacecraft_manipulator_3dof robot model.
[ INFO] [1574075420.948812580]: Setting Param Server with Robot Description
[ INFO] [1574075420.960186960]: Robot semantic model successfully loaded.
[ INFO] [1574075420.960321918]: Setting Param Server with Robot Semantic Description
[ INFO] [1574075421.044189287]: Loading robot model 'free_floating_spacecraft_manipulator_3dof'...
[ INFO] [1574075421.044252246]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1574075422.285903776]: Stereo is NOT SUPPORTED
[ INFO] [1574075422.286002734]: OpenGl version: 4.5 (GLSL 4.5).
[ INFO] [1574075422.926427508]: Loading robot model 'free_floating_spacecraft_manipulator_3dof'...
[ INFO] [1574075422.926521966]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1574075423.295881582]: Loading Setup Assistant Complete
*** stack smashing detected ***: /home/ar0058/catkin_ws/devel/lib/moveit_setup_assistant/moveit_setup_assistant terminated
================================================================================REQUIRED process [moveit_setup_assistant-2] has died!
process has died [pid 54460, exit code -6, cmd /home/ar0058/catkin_ws/devel/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/home/ar0058/.ros/log/f893d60e-09f3-11ea-89ab-6c2b59d30063/moveit_setup_assistant-2.log].
log file: /home/ar0058/.ros/log/f893d60e-09f3-11ea-89ab-6c2b59d30063/moveit_setup_assistant-2*.log
Initiating shutdown!
================================================================================
[moveit_setup_assistant-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
I tried the same in ROS Melodic and I was able to generate the collision matrix. So I guess this is something specific to ROS Kinetic. I hope the above is what you are looking for

rhaschke, 2019-11-18:
My primary question was: Can you reproduce this issue each and every time you run MSA?
If so, it would be great if you could compile from source again using these additional config options:
`catkin config --cmake-args -DCMAKE_CXX_FLAGS="-fsanitize=address -fno-omit-frame-pointer -O1"`

This should output some more detailed infos regarding the crash.

shyamashi, 2019-11-18:
Built MoveIt once gain with 
``` 
catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release --cmake-args -DCMAKE_CXX_FLAGS="-fsanitize=address -fno-omit-frame-pointer -O1"

 ```
I can only see one difference in the xacro warning and every time I run in my system, the MSA crashes. 
Pls see the output below. 

``` 
ar0058@clws00083:~/catkin_ws_3dof$ roslaunch moveit_setup_assistant setup_assistant.launch 
... logging to /home/ar0058/.ros/log/651facac-0a04-11ea-89ab-6c2b59d30063/roslaunch-clws00083-92986.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://clws00083:37283/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

auto-starting new master
process[master]: started with pid [92996]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 651facac-0a04-11ea-89ab-6c2b59d30063
process[rosout-1]: started with pid [93009]
started core service [/rosout]
process[moveit_setup_assistant-2]: started with pid [93012]
[ERROR] [1574082468.551708545]: poll failed with error Interrupted system call
[ERROR] [1574082468.555173465]: poll failed with error Interrupted system call
[ERROR] [1574082468.555610757]: poll failed with error Interrupted system call
[rospack] Error: no package given
[librospack]: error while executing command
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
[ INFO] [1574082477.975608549]: Loaded free_floating_spacecraft_manipulator_3dof robot model.
[ INFO] [1574082477.975743133]: Setting Param Server with Robot Description
[ INFO] [1574082477.987753228]: Robot semantic model successfully loaded.
[ INFO] [1574082477.987802770]: Setting Param Server with Robot Semantic Description
[ INFO] [1574082478.047244619]: Loading robot model 'free_floating_spacecraft_manipulator_3dof'...
[ INFO] [1574082478.047370619]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1574082478.882230389]: Stereo is NOT SUPPORTED
[ INFO] [1574082478.882332056]: OpenGl version: 4.5 (GLSL 4.5).
[ INFO] [1574082479.335767621]: Loading robot model 'free_floating_spacecraft_manipulator_3dof'...
[ INFO] [1574082479.335802747]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1574082479.575984897]: Loading Setup Assistant Complete
*** stack smashing detected ***: /home/ar0058/catkin_ws_3dof/devel/lib/moveit_setup_assistant/moveit_setup_assistant terminated
================================================================================REQUIRED process [moveit_setup_assistant-2] has died!
process has died [pid 93012, exit code -6, cmd /home/ar0058/catkin_ws_3dof/devel/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/home/ar0058/.ros/log/651facac-0a04-11ea-89ab-6c2b59d30063/moveit_setup_assistant-2.log].
log file: /home/ar0058/.ros/log/651facac-0a04-11ea-89ab-6c2b59d30063/moveit_setup_assistant-2*.log
Initiating shutdown!
================================================================================
[moveit_setup_assistant-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

My command line history
``` 
 1684  cd catkin_ws_3dof/
 1685  rosdep update
 1686  sudo apt-get update
 1687  sudo apt-get dist-upgrade
 1688  sudo apt-get install python-wstool python-catkin-tools clang-format-3.9
 1689  source /opt/ros/kinetic/setup.bash
 1690  wstool init src
 1691  wstool merge -t src https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall
 1692  wstool update -t src
 1693  rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}
 1694  catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release --cmake-args -DCMAKE_CXX_FLAGS="-fsanitize=address -fno-omit-frame-pointer -O1"
 1695  catkin build
 1696  git clone https://github.com/shyamashi/free_floating_3_dof_arm.git
 1697  catkin build
 1698  roslaunch moveit_setup_assistant setup_assistant.launch 
 1699  history
```

rhaschke, 2019-11-18:
I was hoping that the instrumented code would produce a backtrace. That's not the case.
Could you please run MSA manually via gdb:
`gdb catkin_ws_3dof/devel/lib/moveit_setup_assistant/moveit_setup_assistant`
Inside `gdb`: run <Enter>
If it crashes, please post the resulting stacktrace (print it with `bt`).

shyamashi, 2019-11-19:
When run MSA as ``` gdb catkin_ws_3dof/devel/lib/moveit_setup_assistant/moveit_setup_assistant ```, it failed to respond after I clicked Generated Collision Matrix. I had to force quit it. The backtrace is

```
(gdb) bt
#0  0x00007ffff4dc1428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff4dc302a in __GI_abort () at abort.c:89
#2  0x00007ffff4e037ea in __libc_message (do_abort=do_abort@entry=1, 
    fmt=fmt@entry=0x7ffff4f1b49f "*** %s ***: %s terminated\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff4ea515c in __GI___fortify_fail (msg=<optimised out>, 
    msg@entry=0x7ffff4f1b481 "stack smashing detected") at fortify_fail.c:37
#4  0x00007ffff4ea5100 in __stack_chk_fail () at stack_chk_fail.c:28
#5  0x00007fffe20bb719 in refinePortal () from /opt/openrobots/lib/libccd.so.2
#6  0x00007fffe20bb863 in ccdMPRPenetration () from /opt/openrobots/lib/libccd.so.2
#7  0x00007fffe712215f in fcl::details::GJKCollide(void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*), unsigned int, double, fcl::Vec3fX<fcl::details::Vec3Data<double> >*, double*, fcl::Vec3fX<fcl::details::Vec3Data<double> >*) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#8  0x00007fffe6efb8cc in ?? () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#9  0x00007fffe6efbad6 in fcl::MeshShapeCollisionTraversalNodeOBBRSS<fcl::Cylinder, fcl::GJKSolver_libccd>::leafTesting(int, int) const () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#10 0x00007fffe7131786 in fcl::collisionRecurse(fcl::CollisionTraversalNodeBase*, int, int, std::__cxx11::list<fcl::BVHFrontNode, std::allocator<fcl::BVHFrontNode> >*) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#11 0x00007fffe7131786 in fcl::collisionRecurse(fcl::CollisionTraversalNodeBase*, int, int, std::__cxx11::list<fcl::BVHFrontNode, std::allocator<fcl::BVHFrontNode> >*) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#12 0x00007fffe6f70f67 in unsigned long fcl::details::orientedBVHShapeCollide<fcl::MeshShapeCollisionTraversalNodeOBBRSS<fcl::Cylinder, fcl::GJKSolver_libccd>, fcl::OBBRSS, fcl::Cylinder, fcl::GJKSolver_libccd>(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::GJKSolver_libccd const*, fcl::CollisionRequest const&, fcl::CollisionResult&) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#13 0x00007fffe714b260 in unsigned long fcl::collide<fcl::GJKSolver_libccd>(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::GJKSolver_libccd const*, fcl::CollisionRequest const&, fcl::CollisionResult&) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#14 0x00007fffe714af33 in fcl::collide(fcl::CollisionObject const*, fcl::CollisionObject const*, fcl::CollisionRequest const&, fcl::CollisionResult&) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#15 0x00007fffed6fdc48 in collision_detection::collisionCallback(fcl::CollisionObject*, fcl::CollisionObject*, void*)
    () from /home/ar0058/catkin_ws_3dof/devel/.private/moveit_core/lib/libmoveit_collision_detection_fcl.so.1.0.1
#16 0x00007fffe73a2242 in fcl::details::dynamic_AABB_tree::collisionRecurse(fcl::NodeBase<fcl::AABB>*, fcl::NodeBase<fcl::AABB>*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*)) ()
   from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#17 0x00007fffed72925f in collision_detection::CollisionEnvFCL::checkSelfCollisionHelper(collision_detection::CollisionRequest const&, collision_detection::CollisionResult&, moveit::core::RobotState const&, collision_detection::AllowedCollisionMatrix const*) const ()
   from /home/ar0058/catkin_ws_3dof/devel/.private/moveit_core/lib/libmoveit_collision_detection_fcl.so.1.0.1
#18 0x00007ffff43c4c9f in moveit_setup_assistant::computeDefaultCollisions[abi:cxx11](std::shared_ptr<planning_scene::PlanningScene const> const&, unsigned int*, bool, unsigned int, double, bool) ()
   from /home/ar0058/catkin_ws_3dof/devel/.private/moveit_setup_assistant/lib/libmoveit_setup_assistant_tools.so.1.0.1
#19 0x00007ffff6b41d6e in moveit_setup_assistant::DefaultCollisionsWidget::generateCollisionTable(unsigned int*) ()
   from /home/ar0058/catkin_ws_3dof/devel/.private/moveit_setup_assistant/lib/libmoveit_setup_assistant_widgets.so.1.0.1
#20 0x00007ffff6b54aef in boost::detail::thread_data<boost::_bi::bind_t<boost::_bi::unspecified, boost::function<void (unsigned int*)>, boost::_bi::list1<boost::_bi::value<unsigned int*> > > >::run() ()
   from /home/ar0058/catkin_ws_3dof/devel/.private/moveit_setup_assistant/lib/libmoveit_setup_assistant_widgets.so.1.0.1
#21 0x00007ffff1adc5d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#22 0x00007ffff4b766ba in start_thread (arg=0x7fffcd4cb700) at pthread_create.c:333
#23 0x00007ffff4e9341d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) 
```

rhaschke, 2019-11-19:
Thanks for providing the backtrace. Am I correct that you have build MoveIt `master`, because the backtrace refers to `CollisionEnvFCL` which was introduced only recently in #1584.
@j-petit: Could you have a look a this? Although MSA runs the collision checks in a separate worker thread, I don't believe this is a race condition issue, because there is only one thread accessing the collision checker.

@shyamashi, can you please confirm your build environment:
Ubuntu 16.04, ROS Kinetic, FCL version?

j-petit, 2019-11-20:
I'll look into this today. However, the original issue created on August 2nd was before the unified collision environment was merged.

rhaschke, 2019-11-20:
You are right. I only have looked at the commit date originally... Probably, the underlying reason is another one then. Thanks for looking into this.

j-petit, 2019-11-20:
I don't have a machine available with 16.04 + Kinetic. Is there a moveit docker image or are all of them now based on 18.04?

shyamashi, 2019-11-20:
I am still using Ubuntu 16.04, ROS Kinetic, libfcl-0.5-dev. 
As I said before, MSA works fine on Ubuntu 18, ROS melodic  and I am thinking of upgrading in a day or two. Thanks

rhaschke, 2019-11-20:
Closing as won't fix in Kinetic and not reproducible in Melodic.

closed 2019-11-20 by rhaschke

akmandor, 2022-04-25:
Issue:
When generating collision matrix in MoveIt Setup Assistant (MSA) using the URDF which contains colliding link geometries, the "stack smashing detected" error happens and the MSA terminates. It would be really helpful if you could update the MSA to resolve this issue.

Although, you can advocate that URDF should be prepared without any geometric overlap, I also think that "The URDF specs don't state that collision geometry cannot overlap" as stated in [this comment](https://github.com/ros-industrial/universal_robot/issues/597#issuecomment-1108098354) for the same issue.

Tests:
- The issue is experienced with the binary and source (master branch) installation of MoveIt.
- The issue is experienced when both creating new or editing the existing MoveIt configuration packages.
- The issue is not experienced when the link geometries in URDF do not collide. For example the default URDF of [jackal robot](https://github.com/jackal/jackal/blob/noetic-devel/jackal_description/urdf/jackal.urdf.xacro) have the same issue but when the "track" property is changed from "0.37559" to "0.38" (which slightly moves the tire way from the main chassis), the collision matrix can be successfully generated using the Setup Assistant.
- The issue is experienced with using the URDF of [Universal robot arms (URDF files: ur{3,5,10}{-.e}_robot.urdf.xacro)](https://github.com/ros-industrial/universal_robot)
- The issue is not experienced using the (unofficial) URDF of another robot arm, Franka Emika's Panda robot](https://github.com/justagist/franka_panda_description/blob/master/robots/panda_arm.urdf.xacro). However, when the official URDF of the same robot arm is used in [franka_ros](https://github.com/frankaemika/franka_ros/blob/develop/franka_description/robots/panda_arm.urdf.xacro), the same error is happened.

System details:
- Ubuntu 20.04 with ROS Noetic (desktop full) installation