moveit/moveit #1597: Consider using #pragma once, instead of include guards

davetcoleman, 2019-08-02:
While reviewing the [pending TrajOpt PR](https://github.com/ros-planning/moveit/pull/1593/files) I noticed some copied code from [trajopt_ros](https://github.com/ros-industrial-consortium/trajopt_ros) used ``#pragama once``. I was about to request its removal, but really, why bother? 

I did some research and found a very useful rational from the Drake Motion Planning project in this [issue](https://github.com/RobotLocomotion/drake/issues/2104). After very informed debate, they decided to break away from the Google Style guide here. Please review that rational so we don't have to reiterate the exact same discussion.

Overall, Drake and TrajOpt use ``pragma once`` to simplify the code and reduce user error. I think MoveIt should too.

davetcoleman, 2019-08-07:
@rhaschke @mlautman @henningkayser thoughts?

rhaschke, 2019-08-07:
I'm advocating `#pragma once` for long time already...

v4hn, 2019-08-09:
:+1: Feel free to add a pull-request for it.

jonbinney, 2019-08-13:
I used this tool to convert Iron Ox's code base over to pragma once: https://github.com/marcusmueller/include-guard-convert

I went ahead and used it to convert moveit over; PR here: https://github.com/ros-planning/moveit/pull/1615

closed 2019-08-13 by rhaschke

davetcoleman, 2019-08-13:
Whoot thanks!