moveit/moveit #1612: rename jog_arm to moveit_jog_arm

v4hn, 2019-08-12:
See https://github.com/ros-planning/moveit/pull/1606#discussion_r312545808 .

If anyone is not happy with the request to rename it, we can discuss here.

AndyZe, 2019-08-12:
I'm slightly not-in-favor of this change. `jog_arm` is so much easier to remember/type than `moveit_jog_arm`.

Also, are we going to rename the other packages in moveit_experimental? Those names will be really long.

moveit_collision_distance_field_ros
moveit_kinematics_constraint_aware
etc

rhaschke, 2019-08-12:
That's the way it is. We also have `moveit_ros_planning_interface`...

v4hn, 2019-08-13:
> moveit_collision_distance_field_ros
> moveit_kinematics_constraint_aware
> etc

Actually, all the "catkinized" packages in there already include the `moveit_` prefix.
There is
- `set(MOVEIT_LIB_NAME moveit_kinematics_cache)`
- `set(MOVEIT_LIB_NAME moveit_kinematics_constraint_aware)`

PR #1663 (2019-09-19): Rename jog_arm->moveit_jog_arm
Closes #1612
closed 2019-09-19 by davetcoleman