moveit/moveit #1616: Unable to jog out of singularity

SansoneG, 2019-08-13:
### Description

When using jog_arm_server with delta_jog_cmds it is possible to jog into a singularity and halt the robot, but then it is not possible to jog out of the singularity again.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source build
* git commit 5199813dc7359cd56c4c6dcf31789423a07f3284

### Steps to reproduce
1. Start Robot in (almost) singular position
[image]
2. Try jogging by publishing on /jog_arm_server/delta_jog_cmds, rotation around x-axis
[image]

### Expected behaviour
As per function `JogCalcs::decelerateForSingularity` in `jog_calcs.cpp:503` it should be possible to jog away from a singularity, because **direction** towards as well as distance to singularity is taken into account.

### Actual behaviour
It is not possible to jog away from the singularity, as seen from rqt_plot provided. Rotating around the X-Axis should (in theory) move the robot out of singularity. Tried rotating in negative and positive direction, if anything it only gets closer to singularity.

SansoneG, 2019-08-13:
### Update

It is possible to get the Robot out of that stuck position by rotating around the Y-Axis, which first causes Joint 4 to move down and then Joint 5 to move out of singularity.

Notice Joint 4 moving down:
[image]

After that i tried rotating around the Y-Axis in reverse, but the robot did not move past the singularity (which should be intended behaviour I guess):
[image]

Question still remains why it wasn't possible to rotate out of the singularity through the X-Axis from the original position though.
Since the input for jogging is some sort of human input (who is probably not able to achieve such a constant direction/amplitude), it is more improbable that the human can jog out of the singularity.

AndyZe, 2019-08-13:
I know a different way that will probably make this better, so I will try to implement that soon. I can see it's significant

felixvd, 2019-08-27:
Any news about or insights into that different way you mentioned @AndyZe ?

SansoneG, 2019-09-04:
What about the method explained in this paper?
[Singularity Avoidance Control of a Non-Holonomic Mobile Manipulator for Intuitive Hand Guidance](https://www.mdpi.com/2218-6581/8/1/14/htm)
Also see the [video](https://www.mdpi.com/2218-6581/8/1/14/htm#app1-robotics-08-00014)

TLDR:
Make the robot avoid the singularity by defining a threshhold around it. If the user tries to jog into the singularity beyond the threshhold a counterforce, getting stronger the closer to the singularity, pushes the manipulator back to the threshhold.

What do you think @AndyZe?

AndyZe, 2019-09-04:
Well I'm willing to test anything with an open mind. Ideally we would have several different modes to choose between.

From skimming that paper, it looks specific to a UR5. For example, this sentence:

> With a fully stretched elbow, the EE looses its ability to move further away from its base and the arm is in a singular configuration. We avoid this by applying a force to the EE with direction back to origin of the base of the serial manipulator whenever the elbow (joint 3) get closer than a specified distance to the critical joint position

That probably won't be extensible to other robots.

You can do some other tricks like setting columns of the Jacobian to zero to avoid motion in a specific direction. That's probably a better approach b/c it  is extensible to any serial manipulator.

AndyZe, 2019-09-04:
My preferred way right now would be:

if max(abs(q_dot) > MAX_JOINT_VELOCITY)
q_dot = q_dot / (max(abs(q_dot)) / MAX_JOINT_VELOCITY);