moveit/moveit #1628: Octomap not properly refreshing Intel Realsense SR300 MoveIt

MarcoT11, 2019-08-16:
### Your environment
* ROS Kinetic
* Ubuntu 16.04
* Sensor: Intel Realsense SR300
* sr300_config.yaml
sensors:
  - sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
    point_cloud_topic: /camera/depth/color/points
    max_range: 1.7                                      #Points further than this will not be used
    point_subsample: 1                                #One=choose every point
    padding_offset: 0.04                               #Size of  padding in cm
    padding_scale: 1.0                                 #Scale of the padding
    max_update_rate: 100                            #Update of Octomap: Less than or equal to this value
    filtered_cloud_topic: filtered_cloud_topic

### Steps to reproduce
* Intel SR300 needed
* Camera is positioned 1 meter above the robot arm

### Expected behaviour
Refreshing of the OctoMap after the obstacle is removed.

### Actual behaviour
Voxels still remain in the position where the obstacle was. 

RQT_graph:
[image]

Workspace:
[image]
Put in an obstacle:
[image]
Remove obstacle:
[image]
Voxels stay even obstacle is removed and pointcoud is not detecting. Same again:
[image]
[image]
When putting in an obstacle below the previous one the voxels disappear:
[image]
When pulling an obstacle upwards it creates a pillar: 
[image]
[image]
which can be removed again by putting in an obstacle below it:
[image]

This causes lots of voxels left in the workspace limiting the motion ability of the robot arm. The voxels especially stay on the edge of the sensor
[image]

When restarting MoveIt the Octomap builds up fresh.

### Backtrace or Console output
`[ WARN] [1565962011.410588435]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)`
`[ WARN] [1565962011.490164733]: No stream match for pointcloud chosen texture Process - Color`
`[ERROR] [1565962011.628675414]: Region-of-interest is not implemented for this device!`
 
What influences the update of the Octomap in MoveIt?
Why do some Voxels remain even though no pointcloud data is received for that position?

v4hn, 2019-08-16:
As far as I understood your post, you observe that voxels are not cleared unless points *behind* the voxel (a perceived from the camera's point of view) are detected.
This is actually intended. The system performs [ray tracing](https://github.com/ros-planning/moveit/blob/master/moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp#L300) to avoid just that.
This is usually done in depth-based perception (also for simple 2d laser scans) to account for measurement errors where no distance is perceived for a ray or something else suddenly blocks the view of a previously perceived obstacle.

Of course, you can hack the plugin to free *all* cells that are not occupied and you will not see the "tower" artifacts anymore.
I believe such an updater might even be welcome upstream.
However, it would throw away information about potential collisions in valid scenarios too.
E.g., just because a robot arm moved over the table, that does not mean there is no table anymore.
So it will never become the default behavior.

PS: If applicable, you should use the [depth_image_octomap_update](https://github.com/ros-planning/moveit/tree/master/moveit_ros/perception/depth_image_octomap_updater) instead of the pointcloud-based updater.

MarcoT11, 2019-08-21:
Hello v4hn, thanks for the answer. What are the advantages of the depth image over the point cloud? 
As far as I know the depth image has a way smoother and more complete stream since it takes more time to optimize the stream of points into an image. The raw point cloud on the other hand is less visual smooth but offers a very quick input of data. 
Wouldn't be a quick data input more important for the field of obstacle avoidance? The faster an obstacle is detected the better? 
Or is the time difference between an depth image and the point cloud so small making the discussion irrelevant?

closed 2019-08-31 by MarcoT11

ChangcongWang, 2021-05-03:
Hello Marco, I want to ask you how you solve this problem finally. I face the same problem but I don't know how to hack the plugin.

ChangcongWang, 2021-05-03:
I figure out how to solve this problem, just need to add "tree_->clear()" to [here](https://github.com/ros-planning/moveit/blob/39f592037576d5fada0e0b64c27370222c7c286c/moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp#L332) to clear the tree created before. I hope it's helpful for who have the same problem.

dbdxnuliba, 2022-12-12:
@ChangcongWang  dou you mean the code should be modified as:
  for (const octomap::OcTreeKey& free_cell : free_cells)
{
     tree_->clear();
      tree_->updateNode(free_cell, false);
}

ChangcongWang, 2022-12-12:
> @ChangcongWang dou you mean the code should be modified as: for (const octomap::OcTreeKey& free_cell : free_cells) { tree_->clear(); tree_->updateNode(free_cell, false); }

No. I mean 
```
try
  {
    tree_->clear();
    /* mark free cells only if not seen occupied in this cloud */
    for (const octomap::OcTreeKey& free_cell : free_cells)
      tree_->updateNode(free_cell, false);
```
However, this refresh method is very inefficient, and the tree must be refreshed every time the tree is updated. At present, I can't find a better way to solve the refresh problem of octomap.

dbdxnuliba, 2022-12-13:
yes it is, we need find a efficient way
and ideas @rojkov @rhaschke