moveit/moveit #1633: move_group.plan() returns 'valid' plan while attached gripper starts in collision state

PaulRuvu, 2019-08-20:
### Description
Hi,

I have a robotic arm with a gripper attached. This gripper is represented by a collision body (box) in my robot URDF. When I request a plan from move_group to a set of joint targets, while my gripper is in collision, it returns a valid plan. 
In all other cases move_group does incorporate the gripper in it's collision avoidance.

Remarkable detail: When planning from a state where a robot link (mesh) is in collision, the planner returns None (as expected).  

Start state:
[image]

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build, tag: 0.9.17

### Steps to reproduce
1. Spawn a robot with a collision box (gripper) attached to the end link in an environment with a second collision object present (mesh in my case). 

2. Move the gripper box into the collision object using  `compute_cartesian_path()` with `avoid_collisions=False`.

3. Set joint targets using `move_group.set_joint_value_target()`

4. Plan a trajectory to the target joints using `move_group.plan()`

### Expected behaviour
`move_group.plan()` should return None since it should detect a collision.

### Actual behaviour
It return a plan as if the gripper does not exist.

v4hn, 2019-08-20:
Does this happen for slight collisions only or also for more severely overlapping meshes?

It is actually a common problem in MoveIt that people expect the robot to move, although the arm
is in *slight* collision with the environment, e.g., on top of a table.
Thus, there is [a planning request adapter](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_request_adapter_plugins/src/fix_start_state_collision.cpp) in place that automatically attempts to resolve such "minor" collisions.
Maybe your case is "fixed" by this adapter?

PaulRuvu, 2019-08-21:
> Does this happen for slight collisions only or also for more severely overlapping meshes? It is actually a common problem in MoveIt that people expect the robot to move, although the arm is in [...]

Thanks for the reply. It happens for severe collisions (as depicted above) where the gripper (in blue) completely penetrates the collision object (the white plane). 
In contrary to the problem you describe, I do find a plan while I am expecting the planner to fail. So I am afraid this adapter does not solve my issue.

v4hn, 2019-08-21:
Ah, now I understand the screenshot you posted. This looks like a severe collision indeed.

Does the resulting planned trajectory also remain in collision for the first few waypoints?
Does the `check_state_validity` rosservice consider this start-state valid?

Also, which version of FCL are you using? I have seen bugs with table-style collisions with older version, so you might want to compile their `master` branch [from source](https://moveit.ros.org/install/source/dependencies/).

PaulRuvu, 2019-08-22:
>  Does the check_state_validity rosservice consider this start-state valid?

According to the  `check_state_validity` service the robot state where the gripper is in collision is invalid (as is expected). However `move_group.plan()` does return a trajectory.

>  Does the resulting planned trajectory also remain in collision for the first few waypoints? 

Interesting: The first waypoint equals the robot's start-state, the second waypoint jumps to a valid state, from there on the trajectory is smooth again. The first two waypoints of this trajectory are depicted below, notice the large position jumps.

[image]

> Also, which version of FCL are you using? I have seen bugs with table-style collisions with older version, so you might want to compile their master branch from source.

I found this issue using both MoveIt's kinetic-devel branch (version 0.9.17)  and melodic-devel branch (version 1.0.2-alpha). I'm not sure how to find my FCL version. MoveIt's master branch does not compile for me on Kinetic nor Melodic, not yet sure why.

rhaschke, 2019-08-22:
For MoveIt's master branch, you probably need to include latest moveit_msgs as well.

v4hn, 2019-08-22:
not just `moveit_msgs`, [see the corresponding rosinstall](https://github.com/ros-planning/moveit/blob/master/moveit.rosinstall)

@PaulRuvu This looks very much like the planning request adapter which should not be active "repairs" the start state and planning continues from this (second) waypoint.
[The original start state is then added back to the resulting plan](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_request_adapter_plugins/src/fix_start_state_collision.cpp#L162).
Please experiment with [the three corresponding parameters](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_request_adapter_plugins/src/fix_start_state_collision.cpp#L194) to see if you can fix this..

PaulRuvu, 2019-08-27:
Thanks a lot! It was indeed the effect of this adapter. Since I don't need it for this application I just set the `jiggle_fraction` to 0 which solved the issue. 

Perhaps a smaller default value would be desired since the purpose of the adapter is to find solutions for "slight" collisions, not like mine as described above.

v4hn, 2019-08-27:
The `jiggle_fraction` parameter is given as a fraction of the range of each joint with a default of `0.02`.
For the "standard" example of a `2*pi` range this means each joint is allowed to "wiggle" by `0.125` radian.
In general it's hard to give good default values for these parameters, but this seems like a lot indeed and I would support a pull-request to reduce the default at least to `0.01`

closed 2019-08-27 by PaulRuvu