moveit/moveit #1641: Dropping `const` of the collision checking process

j-petit, 2019-08-22:
All collision checking functions are declared as `const` in MoveIt. This enables the whole planning scene to be `const` during planning and collision checking.

However, for Bullet and the unified collision environment seems better to not have them `const` for the following reasons:
- speed: `const` requires cloning of the bullet collision managers as we are not allowed to manipulate the persistent member collision manager. This induces large overhead which slows Bullet considerably down. Bullet will not be faster than FCL in most cases except when the closest distance is needed.
- clarity: The necessary cloning makes the collision code in many places obscure. Also in-built Bullet functions for fast broadphase checking can not be used as the broadphase has to be built up again after the clone. 

Dropping `const` from the collision checking process is not easy. It changes the API throughout the codebase as the planning scene (of which the collision environment is a member) will also not be `const`. I spent roughly 1 day trying to get rid of the `const` and still run into compilation errors. So far changes are `19 files changed, 196 insertions(+), 226 deletions(-)`. See [this branch](https://github.com/j-petit/moveit/tree/dropping-const) for details.

Another option would be to declare the Bullet collision managers as `mutable` although this could be considered bad software practice.

j-petit, 2019-08-22:
In the maintainers meeting, `mutable` was favored and will be used for the Bullet collision managers.

closed 2019-08-22 by j-petit

davetcoleman, 2019-08-22:
^ I was just writing that!

However, could you create a unit test that proves that multi-threading is still supported with mutable?

j-petit, 2019-08-23:
It's on my list, I'll try to come up with something.

j-petit, 2019-09-03:
Added PR #1657 for multi-threaded collision testing. However, multi-threading is not working with Bullet due to the `mutable`. I will explain it more in-depth in the future work issue #1646 .