moveit/moveit #1643: Boost.Python.ArgumentError: Python argument types in MoveGroupInterface.set_named_target(MoveGroupInterface, Pose) did not match C++ signature.

anubhav1772, 2019-08-24:
I am fairly new to ROS. I am following [this](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html) tutorial. While trying to plan a motion for my move group to a desired pose using below code(given [here](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html#planning-to-a-pose-goal))
```
pose_goal = geometry_msgs.msg.Pose()
pose_goal.orientation.w = 1.0
pose_goal.position.x = 0.4
pose_goal.position.y = 0.1
pose_goal.position.z = 0.4
move_group.set_pose_target(pose_goal)
```
I am getting below error message:
```
group.set_named_target(pose_target)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 325, in set_named_target
    if not self._g.set_named_target(name):
Boost.Python.ArgumentError: Python argument types in
    MoveGroupInterface.set_named_target(MoveGroupInterface, Pose)
did not match C++ signature:
    set_named_target(moveit::planning_interface::MoveGroupInterfaceWrapper {lvalue}, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)
```
###  My environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Python version: 2.7

I have gone through [issues#842](https://github.com/ros-planning/moveit/issues/842) and was able to understand that this is happening because C++ signature expects a `string` as opposed to an argument of type `Pose(geometry_msgs.msg.Pose)`, but still couldn't figure out a way to solve this. I installed the ros moveit package using `sudo apt-get install ros-melodic-moveit` command. I would really appreciate if someone guide me on this. Thank you so much.

BryceStevenWilley, 2019-08-26:
> group.set_named_target(pose_target)

Where is this line that's causing the error? A quick search doesn't show that particular line anywhere in the moveit codebase, is it a part of your code? If it is, then you've mentioned the problem yourself: `pose_target` should be a string and isn't.  If that line isn't in your code, could you give us more information about your install, i.e. your moveit version? (`rosversion moveit_commander/moveit_ros` would get you that).

anubhav1772, 2019-08-27:
You are right that the parameter to `set_named_target()` function should be string and is working fine if passing the name of an already defined pose, like "home" (which is a string), but I was just trying to use the code from the official melodic-moveit documentation, [Planning to a Pose Goal](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html#planning-to-a-pose-goal). I think this might be an error.

anubhav1772, 2019-08-28:
My bad. Thank you so much for pointing out my mistake.

closed 2019-08-28 by anubhav1772