moveit/moveit #1644: fatal error: Eigen/Geometry: No such file or directory #include <Eigen/Geometry>

anubhav1772, 2019-08-25:
I am fairly new to ROS (melodic), and I am writing a C++ program to use the MoveIt! package. On building my package using `catkin_make`, I am getting below error:

```
In file included from /opt/ros/melodic/include/moveit/robot_model/joint_model_group.h:41:0,
                 from /opt/ros/melodic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/melodic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/anubhav1772/catkin_ws/src/robot_moveit_config/src/robot_manipulation/robot_state.cpp:5:
/opt/ros/melodic/include/moveit/robot_model/joint_model.h:47:10: fatal error: Eigen/Geometry: No such file or directory
 #include <Eigen/Geometry>
          ^~~~~~~~~~~~~~~~
compilation terminated.
```
After going through few documentation online, I found out that I forgot to add below lines in my `CMakeLists.txt` file:
```
find_package(Eigen REQUIRED)
include_directories(include  ${Boost_INCLUDE_DIR} ${EIGEN_INCLUDE_DIRS})
```
But now, I am getting a new error:
```
CMake Error at robot_moveit_config/CMakeLists.txt:14 (find_package):
  By not providing "FindEigen.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen" with any of
  the following names:

    EigenConfig.cmake
    eigen-config.cmake

  Add the installation prefix of "Eigen" to CMAKE_PREFIX_PATH or set
  "Eigen_DIR" to a directory containing one of the above files.  If "Eigen"
  provides a separate development package or SDK, be sure it has been
  installed.
```
I don't have know what to do next because in order to solve above error, everyone seems to add the same two lines(that I told earlier) in the `CMakeLists.txt` on all the discussion forums. Your guidance will be highly appreciated. Thank you so much!

rhaschke, 2019-08-25:
You need to search for Eigen**3**:
`find_package(Eigen3 REQUIRED)`

closed 2019-08-25 by rhaschke

anubhav1772, 2019-08-25:
Thank you for your prompt response. I tried this, but unfortunately it's also not working. I finally solved this error by adding `moveit_ros_planning_interface` in `find_package() `inside `CMakeLists.txt file.`

rhaschke, 2019-08-25:
Sorry, of course you need to use the corresponding **EIGEN3** includes as well:
`include_directories(SYSTEM ${EIGEN3_INCLUDE_DIRS})`

LOKAKSHAI, 2022-03-22:
I'm getting this error

/opt/ros/melodic/include/moveit/robot_model/joint_model.h:47:10: fatal error: Eigen/Geometry: No such file or directory
 #include <Eigen/Geometry>
          ^~~~~~~~~~~~~~~~
compilation terminated.

rhaschke, 2022-03-22:
Obviously, you are missing upstream dependencies. Did you follow the [build instructions from source](https://moveit.ros.org/install/source/) and install dependencies:
`rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}`

hoab02, 2024-05-31:
Sorry, I have the same error but I'm using ROS Noetic, I try to find CMakelist.txt to add two lines you suggested but still have the same error. How can I deal with this problem?