moveit/moveit #1654: Segmentation fault during planning: __memmove_avx_unaligned_erms()

ssnover95x, 2019-08-30:
### Description

Segmentation fault while planning.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build?
Binary: v1.0.2

### Steps to reproduce
I'm having trouble reproducing this behavior consistently, it occurred while planning a move to a joint goal that it had already completed before once while running the program.

Here's a stack trace:
```
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:423
#1  0x00007f6e5ae050bd in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<cha
r> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const
&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007f6e5ca85490 in ?? () from /opt/ros/melodic/lib/libmoveit_robot_state.so.1.0.2
#3  0x00007f6e5ca868ec in moveit::core::robotStateToRobotStateMsg(moveit::core::RobotState const&, mo
veit_msgs::RobotState_<std::allocator<void> >&, bool) ()
   from /opt/ros/melodic/lib/libmoveit_robot_state.so.1.0.2
#4  0x00007f6e5d2b1fe3 in moveit::planning_interface::MoveGroupInterface::MoveGroupInterfaceImpl::con
structMotionPlanRequest(moveit_msgs::MotionPlanRequest_<std::allocator<void> >&) ()
   from /opt/ros/melodic/lib/libmoveit_move_group_interface.so.1.0.2
#5  0x00007f6e5d2c232a in moveit::planning_interface::MoveGroupInterface::MoveGroupInterfaceImpl::pla
n(moveit::planning_interface::MoveGroupInterface::Plan&) ()
   from /opt/ros/melodic/lib/libmoveit_move_group_interface.so.1.0.2
```

rhaschke, 2019-08-30:
Could you please reproduce this issue with a debug-build from latest melodic source?

suryakanths, 2024-11-14:
@ssnover95x have you solved this issue?