moveit/moveit #1655: Class default_planner_request_adapters/ResolveConstraintFrames does not exist

rickstaa, 2019-08-31:
### Description

While following the [moveit_tutorials](https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) I get an error while trying to run the `roslaunch panda_moveit_config demo.launch` command.

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build: Binary
* If binary, which release version?:  1.0.2-1bionic.20190808.234100 amd64

### Steps to reproduce
Install moveit and perform the steps in the [moveit_tutorials](https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html).

### Expected behaviour
No error.

### Actual behaviour
Error:

```
[ERROR] [1567265168.348453066]: Exception while loading planning adapter plugin 'default_planner_request_adapters/ResolveConstraintFrames': According to the loaded plugin descriptions the class default_planner_request_adapters/ResolveConstraintFrames with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  chomp/OptimizerAdapter default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds
```

## Possibly related Issues
- https://github.com/ros-planning/moveit/issues/1452
- https://github.com/xqms/rosmon/issues/16

rhaschke, 2019-08-31:
> Class `default_planner_request_adapters/ResolveConstraintFrames` does not exist

This plugin was added to the `master` branch of MoveIt, which is not released as binary debs.
Hence, you need to build from source.

The tutorials you referenced correspond to the master branch.
When running against stock Melodic MoveIt you should use the [Melodic tutorials](http://docs.ros.org/melodic/api/moveit_tutorials/html/index.html).

> As I want to use a gazebo simulation of the panda robot I use a fork of the panda_moveit_config and ros_franka packages. Does this automatically mean that I have to build the whole moveit package from source?

Usually not. However, on issues first compare the config and launch files in `panda_moveit_config`.

closed 2019-08-31 by rhaschke

huiwenzhang, 2020-06-30:
@rhaschke Hi, even I follow the tutorials of the melodic version, I still encounter this error. However, according to the melodic version tutorial, a binary installation of the moveit is ok. To solve this issue, I can install the master branch of moveit from source, But I am doubting  whether the source install will corrupt somewhere else of tutorial.

rhaschke, 2020-06-30:
You can safely ignore this "error" on a Melodic binary release. As said before, the corresponding plugin is only available in master (to be built from source) and it will become available in Noetic. The corresponding [_subframes_ tutorial](https://ros-planning.github.io/moveit_tutorials/doc/subframes/subframes_tutorial.html) is only available in the master version of the tutorials.