moveit/moveit #1659: optionally access database states via `MoveGroupInterface`

v4hn, 2019-09-06:
Currently, the `MoveGroupInterface` supports a `setNamedTarget` method that works with joint configurations stored in the `srdf` file of the moveit config.

In contrast, the RViz Motion Planning Plugin supports setting targets from the warehouse too.

It would be nice to extend the MGI to support an optional database connection and extend the semantics of `setNamedTarget` to also lookup target states from this database if if the currently implemented logic does not find a valid state associated with the passed string.

felixvd, 2020-10-23:
IMO it would be much nicer if the database access was abstracted to a class which can either connect to MongoDB or to a file in the `moveit_config`'s `config` folder. That file could hold both the saved joint configurations that are currently in the `srdf`, as well as `PoseStamped` targets, which are currently extremely inconvenient to record.

My reasoning is that practically no one seems to use the warehouse feature, or at least no one who has updated the tutorials in the last 2+ years. Related: https://github.com/ros-planning/moveit_tutorials/issues/139, https://github.com/ros-planning/warehouse_ros/issues/42  
Using a local file would be significantly less complicated and cover the majority of use cases, I believe (but I am biased since I don't do logistics).

Would like to ask @RoboticsYY to weigh in since he seems to be porting the warehouse to ros2.

RoboticsYY, 2020-10-24:
I rarely used MongoDB in the past. From my point of view, I feel that MongoDB seems too large for such tasks, usually only a few robot states are saved to the database instead of thousands or millions. Maybe MongoDB is a better choice for some applications that need to record big data (such as AI training) or more suitable for data sharing between multiple robots.

gleichdick, 2020-10-25:
What do you think about a sqlite adapter for warehouse_ros?

gleichdick, 2020-11-05:
I had some spare time: https://github.com/gleichdick/warehouse_ros_sqlite
Any feedback and maybe some help is very welcome :smiley:

v4hn, 2021-06-15:
Sorry, I found this only now.

@gleichdick You actually missed @k-okada's efforts to do the same thing
https://github.com/k-okada/warehouse_ros_sqlite
a year earlier.

I don't know which one is better, but I would still like
to move one of them into the ros-planning organization for visibility
and as the recommended backend for the warehouse.
MongoDB is a joke for most practical applications, though it can be useful with high load/parallel access.

There is also nothing wrong with a warehouse plugin for yaml files either,
but someone would have to write it. ;)
Also at least for robot states this already exists in named states.

gleichdick, 2021-06-17:
Too bad :sweat_smile: 
I just ported my implementation to ROS2, but I need some help to test it with moveit2. Can maybe someone with a working moveit2 environment verify that it's still working?

gleichdick, 2021-06-21:
I'm happy with moving it into ros-planning to make it more visible. The required patches to make it work in ROS2 are ros-planning/warehouse_ros#78, ros-planning/warehouse_ros#79 and ros-planning/moveit2#513.

v4hn, 2021-06-21:
> I'm happy with moving it into ros-planning to make it more visible.
> The required patches to make it work in ROS2 are [...]

The MoveIt2 crew will probably have these merged soon enough.
I would like to see both MoveIt/MoveIt2 supported in an upstream repository.

Before moving it into ros-planning, it would be nice to have a short tutorial available for it to show how to configure and use it. This is a lot to ask for, but it would clearly help advertising the warehouse for actual use.
Would you be willing to contribute a short tutorial?

If you have time, you could also attend [the maintainer meeting this week](https://discourse.ros.org/t/moveit-manipulation-wg-meeting-june-24th/20957) to briefly talk about moveing the repo to our organization.

v4hn, 2021-06-22:
> Thank you for inviting me to the maintainer meeting, I hope I'll be home on time on thursday to drop by.

Let's hope I will actually be around myself. I'm rather notorious this year for putting items on the agenda and then not showing up for family reasons.

> Regarding the tutorial, could you elaborate a bit?

I would expect a short paragraph on how to set it up followed by one or two ways to access information in the warehouse.
Nothing fancy, nothing complicated, but something that helps users to get started if they hear "you can store robot states in sqlite now".

> warehouse_ros does not seem to have ROS2 docs yet..

I would like to have this as part of the MoveIt tutorials, not part of the warehouse_ros docs.
Others might disagree though.

felixvd, 2021-06-23:
I agree that it should be part of the main tutorials. In an ideal world, it would also reference some ways to save/load messages/objects as well (e.g. https://github.com/swri-robotics/message_serialization ).