moveit/moveit #1660: `robot_trajectory::RobotTrajectory::copy` and `::deepcopy`

v4hn, 2019-09-06:
It was just pointed out by me again that
```
robot_trajectory::RobotTrajectory(const RobotTrajectory&)
```
creates only a shallow copy of the trajectory - not copying the underlying `RobotState` waypoints.

This is indeed a common source of frustration and I would propose to address it by changing the interfaces:

- deprecate (and eventually `=delete`) the current copy constructor
- introduce `copy` and `deepcopy` (the names are up for debate) methods

shivaang12, 2019-11-18:
I’ll work on this for WMD 2019

rhaschke, 2019-12-19:
#1760 implements a deepcopy variant of the RobotTrajectory's copy constructor.
There we agreed, that the shallow copy should be kept as the default (https://github.com/ros-planning/moveit/pull/1760#issuecomment-558605283).

closed 2019-12-19 by rhaschke