moveit/moveit #1662: More frequent and synchronized package releases

henningkayser, 2019-09-09:
Some releases of packages in the MoveIt ecosystem don't hold up to the latest changes in MoveIt. For example, moveit_msgs hasn't been released since April even though there are a lot of 
 new contributions in the repository. Also, it would be great if we had a more frequent release cycle (maybe every 1 or 2 weeks) for the main MoveIt repository and keep dependent packages synchronized with the latest version. @rhaschke I or someone else at PickNik would be happy to help setting this up and maintaining this.

gavanderhoorn, 2019-09-09:
> Also, it would be great if we had a more frequent release cycle (maybe every 1 or 2 weeks)

With syncs from staging -> public every once in two to three weeks releasing every week seems a little much.

Unless you want to track the `ros-testing` repository all the time (which may not be such a good idea).

rhaschke, 2019-09-09:
I agree that having more frequent releases would be great. However, settling on a schedule every 1 or 2 months would be sufficient, I think. Preparing and thoroughly! testing a release candidate consumes a lot of time. One major issue with the current situation is that probably every maintainer works on the `master` branch and nobody really tests the `melodic` branch anymore.
However, the `master` branch will only be released into `Noetic`.
So, in order to prepare a release, we should skim through the (bugfix) commits on `master` and (i) list those that should be back-ported into `melodic`, then (ii) actually back-port them, (iii) thoroughly test the release candidate, and (iv) finally actually prepare the release. I'm sure with the last step, @130s will be happy to assist as well.

v4hn, 2019-09-10:
> moveit_msgs hasn't been released since April even though there are a lot of new contributions in the repository

"A lot of new contributions" boils down to subframe support. That's surely enough though.

We talked about release scheduling a lot in a number of maintainer meetings before and
we always end up at "A schedule of 1-2 months between releases would be nice, but someone has to do them".
If PickNik could allocate one of their official maintainers to enforce regular releases (possibly once a month), that would be great.

Aside from the actual release process, nothings stops anyone from creating issues to schedule a new release, listing all the commits/merged pull-requests in later branches, and checking whether they should be backported to `kinetic-devel` and/or `melodic-devel` one by one.
Once everything is backported as appropriate and verified to run, the actual release process can continue.

simonschmeisser, 2019-09-16:
We build, test and use melodic-devel fairly regularly here. I appreciate it being quite stable and think a central point of the master branch is keeping it that way?

v4hn, 2019-09-17:
> We build, test and use melodic-devel fairly regularly here.

That is indeed good to hear and the reason why we keep master and melodic-devel separate.

> I appreciate it being quite stable and think a central point of the master branch is keeping it that way?

Yes. And that is definitely not supposed to change with more frequent releases. However, would you not also appreciate bugfixes, cleanups and light extensions as long as they are API compatible?

Granted, you will have to rebuild your workspaces with each release due to our sonames policy, but I guess everyone deploying MoveIt-based systems should be used to this by now.

simonschmeisser, 2019-09-17:
> Yes. And that is definitely not supposed to change with more frequent releases. However, would you not also appreciate bugfixes, cleanups and light extensions as long as they are API compatible? [...]

Actually as @rhaschke set up some ABI check system for rviz, ABI stability could be a possible goal and then we could stop auto-incrementing the sonames (soversion?). But no, we ourselves do not require it. (This would work better if we had some definition of what parts of the api are public and use pimpl/d-pointers there but that is another issue)

API stability and not introducing new dependencies are more important.

Given the always limited resources I wouldn't put too much effort into the light extensions part.

v4hn, 2019-09-17:
> sonames (soversion?).

soname refers to the technique to use a library-name + some part of the soversion as the linking target.
So you could specify the soname to only include `MAJOR` or `MAJOR.MINOR` while still having `MAJOR.MINOR.PATCH` in the library version.

> Actually as @rhaschke set up some ABI check system for rviz, ABI stability could be a possible goal and then we could stop auto-incrementing the soversion.

We had manual ABI checks for indigo releases when we moved new development to kinetic/lunar. [See here for my semi-automated checker](https://github.com/v4hn/moveit_abi_check).

I'm not a big fan of integrating ABI checks with CI as it adds a lot of build overhead and also can trigger "unreasonable false-positives", e.g., removing some overly specialized internal boost-symbol. (of course if you want to claim this is still part of public ABI, technically you'd be correct)
On the other hand, @rhaschke implemented a caching mechanism with the moveit_ci as far as I know, so that overhead could be halved at least.
@rhaschke, your thoughts would be appreciated.

In order to allow for ABI changing bug-fixes in *all* branches and to avoid even more (non-trivial) overhead with releases, we deliberately chose to have the full soversion as part of the soname and bump it with each release.

At any point in time though, we could freeze the soname for kinetic and/or melodic releases though and manually bump soversions of individual libraries if we actually want to/have to break ABI again (for critical fixes).

> This would work better if we had some definition of what parts of the api are public

One of the big problems with MoveIt is that *everything* is public API.
This is mostly for historical reasons, but it also makes sense with respect to research done using the project.

> ... and use pimpl/d-pointers there but that is another issue)

Feel free to propose pull-requests to add pimpl for relevant classes if you feel that would improve the code base.
Personally, I don't think it matters as long as we are ok with the current soname strategy. :)

Anyway, I guess deep ABI discussions are not the core of this issue, so please file a new issue if you want to follow up on this point.

rhaschke, 2019-09-17:
> I'm not a big fan of integrating ABI checks with CI as it adds a lot of build overhead and also can trigger "unreasonable false-positives".

Our CI is already rather overloaded with time-consuming checks. For example, requiring a full rebuild (e.g. due to changes in basic MoveIt headers) and then performing the clang-tidy checks, usually exhausts the allocated Travis time of 1:30h. In these cases, I had to manually trigger a rebuild in the past. The abi-compliance checker requires roughly 7min for the two rviz libs. Scaling this to MoveIt, we could easily break our Travis time limit again. So, I don't really recommend it on this huge repo.

jschleicher, 2019-10-22:
Is there any progress for at least one next autumn release?

Regarding testing: What about releasing a package right after a melodic sync, so that we can test the buildfarm packages from `testing` for at least two weeks until the next ROS sync?

rhaschke, 2019-10-22:
I won't have resources for this before mid of November.
@jschleicher, which features and bug fixes do you like to see in the next release? `melodic-devel` has only a very few commits since the latest release, while most of the commits directly went to `master` (only).

jschleicher, 2019-10-22:
@rhaschke 669bcaee7dbc5ba0cb722219d780d370f21eed15 and 0d0c399bb4e6929d6937e687e6bf254deffab87b are merged and relevant for me.

davetcoleman, 2019-10-30:
Just seeing and catching up with this thread. We at PickNik are interested in supporting the reoccurring 1-2 month release process for Melodic. I've been mentoring @JafarAbdi to bloom-release all of PickNik's maintained ROS packages outside of MoveIt, and he's become pretty expert at the build farm. Our client projects have needed various ros-planning repos to be released (such as panda_moveit_config for the tutorials) but I wanted to make sure we don't step on any toes of the releases @rhaschke has been doing (and maybe @130s ?). 

Regarding backporting changes from the master branch to the melodic branch, I'd (again) argue we're spending too much time on effort that no one is asking for. If there are Melodic users asking for particular fixes, we should request they do the backport via a PR. Otherwise I think we should let Melodic be "safe" and stable by not increasing the chances we cause a regression in an LTS beyond specific needs.

davetcoleman, 2020-05-06:
I've attempted to address some of these release concerns in this similar issue: https://github.com/ros-planning/moveit/issues/1747

Hopefully we can close this issue soon?