moveit/moveit #1666: Cartesian paths issue

ThanasisTs, 2019-09-12:
Hello,

I am trying to move the end effector of a robotic manipulator through a series of waypoints using MoveIt!. At first I used the move_group.move() command to iterate through each point. The code I used is shown below:

```
ros::init(argc, argv, "manos_openpose");
ros::NodeHandle nh;
ros::AsyncSpinner spinner(1);
spinner.start();
moveit::planning_interface::MoveGroupInterface move_group("arm");

move_group.setPoseReferenceFrame("world");

geometry_msgs::Pose currentPose = move_group.getCurrentPose().pose;

geometry_msgs::Pose target_pose = currentPose;
target_pose.position.z += 0.1;
move_group.setPoseTarget(target_pose);
move_group.move();

target_pose.position.z -= 0.1;
move_group.setPoseTarget(target_pose);
move_group.move();

ros::waitForShutdown();
```

Everything works fine. Then i tried to replicate the same movement by computing a cartesian path passing through these points. The code i used is shown below:

```
ros::init(argc, argv, "manos_openpose");
ros::NodeHandle nh;
ros::AsyncSpinner spinner(1);
spinner.start();
moveit::planning_interface::MoveGroupInterface move_group("arm");

move_group.setPoseReferenceFrame("world");

geometry_msgs::Pose initState = move_group.getCurrentPose().pose;
std::vector<geometry_msgs::Pose> waypoints;

geometry_msgs::Pose target_pose = initState;
target_pose.position.z += 0.1;
waypoints.push_back(target_pose);

target_pose.position.z -= 0.1;
waypoints.push_back(target_pose);

moveit::planning_interface::MoveGroupInterface::Plan my_plan;
moveit_msgs::RobotTrajectory trajectory;
const double jump_threshold = 0.0;
const double eef_step = 0.001;
double fraction = move_group.computeCartesianPath(waypoints, eef_step, jump_threshold, trajectory);

my_plan.trajectory_ = trajectory;
move_group.execute(my_plan);
```

However the function computeCartesianPath() returns a value between 0.3 and 0.6 (different value each time i run the code). Hence, MoveIt! executes only the portion of the trajectory corresponding to the returned value of the above mentioned function. Any ideas why it won't execute the whole trajectory since the waypoints are valid? I am using Ubuntu 16.04 and ROS Kinetic. Thanks a lot in advance and let me know if there is any other information i could provide.

janisa9, 2019-09-20:
Are you sure there are no obstacles in the linear path? If there are, then the value returned by computeCartesianPath() will be below 1.

ThanasisTs, 2019-09-20:
Yes there are no obstacles in the linear path. But even if there were, shouldn't the `move()` function fail as well?

janisa9, 2019-09-20:
Using cartesian path robot TCP will move in a straight line, mby robot is even colliding with itself to move this straight line, in this case, I think it will also return a value below zero. 

with move() motions ar planned to avoid any kind of collisions, it will fail only if there are no possible to find a collision-free path in the given time.

jschleicher, 2019-09-23:
This appears to be a cross post of https://answers.ros.org/question/332783/move_groupmove-vs-cartesian-paths/

Planning in cartesian task space vs. joint space gives different solutions. Please refer to basic robotics literature for path planning, e.g. [Spong: Robot modeling and Control](https://www.researchgate.net/profile/Mohamed_Mourad_Lafifi/post/How_to_avoid_singular_configurations/attachment/59d6361b79197b807799389a/AS%3A386996594855942%401469278586939/download/Spong+-+Robot+modeling+and+Control.pdf).

felixvd, 2019-11-17:
Looks like this can be closed @ThanasisTs ?

ThanasisTs, 2019-11-17:
Indeed, i am closing the issue.

closed 2019-11-17 by ThanasisTs