moveit/moveit #1669: Could not build MoveIT tutorial because: "collision_env_fcl.h: No such file or directory"

EvanLavizadeh, 2019-09-13:
### Description
Could not build tutorial from https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html using "catkin build"

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18
* Source or Binary build? (Of what? Moveit or ROS?)
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Just follow steps from https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html

### Expected behaviour
Build should complete successfully

### Actual behaviour
Build failed

### Backtrace or Console output
https://gist.github.com/EvanLavizadeh/a505fb2a3b311eb4022102b95e50cd2b

EvanLavizadeh, 2019-09-14:
I tried again with the workspace in the home directory instead of my ~/tmp and still errors but different errors seen here:

https://gist.github.com/EvanLavizadeh/602ad53b68f1b434eb3e53e8a1149fa1

j-petit, 2019-09-17:
Did you download the latest source from the master branch?

EvanLavizadeh, 2019-09-18:
> Did you download the latest source from the master branch? Also for all the other repos like `moveit_msgs`? Try to follow the [instructions](https://moveit.ros.org/install/source/) again with fresh sources.

That seemed to work. What was the cause of this?

j-petit, 2019-09-18:
We had major updates throughout the codebase affecting multiple repos and therefore one needs up-to-date sources of all repos to build.

EvanLavizadeh, 2019-09-18:
Ahh okay! Thank you Jens.

closed 2019-09-18 by EvanLavizadeh