moveit/moveit #1670: Failed to import pyassimp.

addy1997, 2019-09-16:
I am using **Ubuntu 16.0.4** with **ROS kinetic**. 

 import moveit_commander
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/vol/sandbox/ros/install/lib/python2.7/dist-packages/moveit_commander/__init__.py", line 3, in <module>
    from planning_scene_interface import *
  File "/vol/sandbox/ros/install/lib/python2.7/dist-packages/moveit_commander/planning_scene_interface.py", line 48, in <module>
    import pyassimp
  File "/usr/lib/python2.7/dist-packages/pyassimp/__init__.py", line 1, in <module>
    from .core import 

Failed to import pyassimp.

Please help me with this.

jschleicher, 2019-09-16:
Looks like https://github.com/ros-planning/moveit/issues/1546, which was only present in the testing version 0.9.16. Which version are you using? `apt` packages or source build? Please install all the dependencies, if you build from source please follow the steps in the [tutorial](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/getting_started/getting_started.html#build-your-catkin-workspace).

addy1997, 2019-09-16:
I am using 0.9.16  version. Also I have installed moveit by building it
from source.

jschleicher, 2019-09-16:
Please upgrade to 0.9.17 (fixed) and install the dependency via rosdep.

addy1997, 2019-09-16:
sure

closed 2019-09-16 by addy1997