moveit/moveit #1673: Contact points are not returned if there is no other collision object

adam-vonderviszt, 2019-09-20:
MoveIt 'check_state_validity' service does not return the contact points of the collision if there is no extra collision object added to the panning place.

If I check a robot state for collision, that has clearly a collision with the urdf environment, the validity result is correct (which is invalid), but the contact points are not returned at all (empty list). But if I add a definitely not colliding extra box to the planning scene far away from the robot with the 'apply_planning_scene' service, I can get the collision contact points of the robot and the urdf environment.
I can hot-fix the issue with adding a dummy-box to the planning scene, but I think this issue needs to be fixed.
Thank you!

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Binary buil, latest release version
* Python

pbeeson, 2020-09-03:
This is still the case in MoveIt master branch as of 09/03/2020.  Unsure if this is by design or not.

v4hn, 2020-09-07:
This is likely an artifact due to code layout. It's definitely not meant to be like that by design.

LoyVanBeek, 2020-09-18:
I think I fixed this in https://github.com/ros-planning/moveit/pull/2319

closed 2021-02-24 by v4hn