moveit/moveit #1677: Setup Assistant says Success! but has loads of terminal errors.

EvanLavizadeh, 2019-09-25:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18
* Source or Binary build? Source
* If source, which git commit or tag?  ff8f1d8

### Steps to reproduce
I can't give exact steps without releasing my urdf files which I can't do but I simply opened the setup assistant in my catkin_ws. I referred to a .urdf file which is in catkin_ws/src/move_it_config/ginger_description/urdf

Then, when I clicked "Load Files," the GUI says "Success! Use the left navigation pane to continue." However, the terminal which I ran the assistant from had loads of errors and warnings. 

### Expected behaviour
I guess expected behavior is to not have any errors.

### Actual behaviour
Terminal errors.

### Backtrace or Console output
https://gist.github.com/EvanLavizadeh/69f688c09a1766c120f8e40754ea901a

rhaschke, 2019-09-25:
The setup assistant doesn't care about concrete meshes. Everything that matters for SA is the kinematic structure, which was successfully loaded. To resolve the errors, you need to advertise the package `ginger_description` in your ROS environment. Probably you forgot to source the corresponding workspace.

closed 2019-09-25 by rhaschke

EvanLavizadeh, 2019-09-26:
@rhaschke 

The reason this concerns me is that rviz (and the rviz panel in the setup assistant) are empty. The robot is not shown. Is this related?