moveit/moveit #1679: MSA: config/sensors_3d.yaml always populated

gavanderhoorn, 2019-09-27:
### Description

The MoveIt Setup Assistant seems to unconditionally write out a `config/sensors_3d.yaml` file, regardless of whether the user has actually selected any of the two options in the *3D Perception* screen (added in #969).

### Your environment
* ROS Distro: Kinetic/Melodic
* OS Version: e.g. Ubuntu 16.04/18.04
* Binaries, versions: `0.9.17`/`1.0.2`

`master` shows the same behaviour.

### Steps to reproduce

 1. start the MSA
 1. create a new configuration package
 1. do not navigate to the *3D Perception* screen (ie: do not click the *3D Perception* button)
 1. navigate to the *Configuration Files* screen
 1. save the configuration

Inspect the contents of `config/sensors_3d.yaml`.

### Expected behaviour
The file should be empty, or should only have the single `sensors` key with an empty list as value:

```yaml
sensors: []
```

### Actual behaviour
The file contains:

```yaml
# The name of this file shouldn't be changed, or else the Setup Assistant won't detect it
sensors:
  - filtered_cloud_topic: filtered_cloud
    max_range: 5.0
    max_update_rate: 1.0
    padding_offset: 0.1
    padding_scale: 1.0
    point_cloud_topic: /head_mount_kinect/depth_registered/points
    point_subsample: 1
    sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
```

gavanderhoorn, 2019-09-27:
Note: the `rosparam` line that actually loads the `.yaml` in `sensor_manager.launch.xml` is not added to that launch file if no sensor option is chosen on the *3D Perception* screen.

Regardless: the `.yaml` should not be generated or be empty.

DLu, 2021-06-04:
Should be fixed by #2708

closed 2021-06-04 by rhaschke