moveit/moveit #1680: MSA: incorrect 3D sensor configuration if no sensor option selected

gavanderhoorn, 2019-09-27:
### Description

The MSA writes out an incorrect `config/sensors_3d.yaml` when a user selects the *None* option on the *3D Perception* screen.

The incorrect yaml causes MoveIt to print an error upon starting any `.launch` file that includes `sensor_manager.launch.xml`.

### Your environment
* ROS Distro: Kinetic/Melodic
* OS Version: e.g. Ubuntu 16.04/18.04
* Binaries, versions: `0.9.17`/`1.0.2`

`master` shows the same behaviour.

### Steps to reproduce

 1. start the MSA
 1. create a new configuration package
 1. navigate to the *3D Perception* screen
 1. make sure the *None* option is selected (the default)
 1. navigate to the *Configuration Files* screen
 1. save the configuration

Inspect the contents of `config/sensors_3d.yaml`.

### Expected behaviour
The file should be empty, or should only have the single `sensors` key with an empty list as value:

```yaml
sensors: []
```

No error should be printed.

### Actual behaviour
The file contains:

```yaml
sensors:
 - {}
```
This is a map, with a sequence (`sensors`), with a single item, an empty map.

This results in an error.

### Backtrace or Console output

As the `sensors` sequence is not empty, the sensor plugins attempt to parse it and fail. This leads to the following error:

```
[ERROR] [..] [/move_group] [ros.moveit_ros_perception]: No sensor plugin specified for octomap updater 0; ignoring.
```

gavanderhoorn, 2019-09-27:
These are probably the problematic lines:

https://github.com/ros-planning/moveit/blob/977d9ad3e64895f5a669f3766f6441b46955b509/moveit_setup_assistant/src/tools/moveit_config_data.cpp#L1000-L1013

notice how the map is always created, as the conditional (is any sensor configured) is only checked after emitting `YAML::BeginMap`.

---

Noting this for myself and potential future others: I don't have the time to submit a PR right now.

gavanderhoorn, 2019-09-28:
ROS Answers post about this: [How to disable moveit octomap](https://answers.ros.org/question/306733).

DLu, 2021-06-03:
Closed by #2708

v4hn, 2021-07-01:
It seems #2708 aggravated the problem instead of resolving the issue.

With `master` right now, there are two different outcomes if you create a new config.

1. If you select "None" in the "3D Perception" tab, you end up with this config:
```
# The name of this file shouldn't be changed, or else the Setup Assistant won't detect it
sensors:
  - {}
  - {}
```

2. If you did not click on the "3D Perception" tab at all (I guess that's what triggers it) and just generate the configuration, you end up with this surprise instead:
```
# The name of this file shouldn't be changed, or else the Setup Assistant won't detect it
sensors:
  - filtered_cloud_topic: filtered_cloud
    max_range: 5.0
    max_update_rate: 1.0
    padding_offset: 0.1
    padding_scale: 1.0
    point_cloud_topic: /head_mount_kinect/depth_registered/points
    point_subsample: 1
    sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
  - far_clipping_plane_distance: 5.0
    filtered_cloud_topic: filtered_cloud
    image_topic: /head_mount_kinect/depth_registered/image_raw
    max_range: 5.0
    max_update_rate: 1.0
    near_clipping_plane_distance: 0.3
    padding_offset: 0.03
    padding_scale: 4.0
    point_cloud_topic: /head_mount_kinect/depth_registered/points
    point_subsample: 1
    queue_size: 5
    sensor_plugin: occupancy_map_monitor/DepthImageOctomapUpdater
    shadow_threshold: 0.2
```