moveit/moveit #1682: Missing large voxels in octomap visualization in RViz

RMonica, 2019-10-02:
### Description

Most large voxels in an octree are not displayed in the planning scene.

The issue is in `moveit_ros/visualization/rviz_plugin_render_tools/src/octomap_render.cpp`, method `octreeDecoding`.
There is an optimization to remove voxels which are completely surrounded by other voxels and can't be seen.
Something is wrong, here: for example, it seems that the `key` variable contains a key for depth `octree_depth_`, but it is used in a full-depth search later: `octree->search(key)`. Changing to `octree->search(key, octree_depth_)` is not enough to fix the issue, however.

If I disable the optimization, like this:
```
diff --git a/moveit_ros/visualization/rviz_plugin_render_tools/src/octomap_render.cpp b/moveit_ros/visualization/rviz_plugin_render_tools/src/octomap_render.cpp
index f427a7315..672b4738b 100644
--- a/moveit_ros/visualization/rviz_plugin_render_tools/src/octomap_render.cpp
+++ b/moveit_ros/visualization/rviz_plugin_render_tools/src/octomap_render.cpp
@@ -172,7 +172,8 @@ void OcTreeRender::octreeDecoding(const std::shared_ptr<const octomap::OcTree>&
     // traverse all leafs in the tree:
     for (octomap::OcTree::iterator it = octree->begin(octree_depth_), end = octree->end(); it != end; ++it)
     {
-      bool display_voxel = false;
+      //bool display_voxel = false;
+      bool display_voxel = ((int)octree->isNodeOccupied(*it) + 1) & render_mode_mask;
 
       // the left part evaluates to 1 for free voxels and 2 for occupied voxels
       if (((int)octree->isNodeOccupied(*it) + 1) & render_mode_mask)
```
the large voxels are displayed.

Example:
With the optimization:
[image]
Optimization disabled:
[image]

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source build: commit 2a3d8f8d4b729b3c7aaf1ed9175df478ff52b5ea

### Steps to reproduce

- Launch RViz.
- Add a planning scene display.
- Publish a planning scene with an octree in which there are voxels bigger than 2x2x2.

I attach my test planning scene [message.txt](https://github.com/ros-planning/moveit/files/3680072/message.txt).
It can be published with
`rostopic pub -f message.txt /planning_scene moveit_msgs/PlanningScene`
The fixed frame is `world`.

MatthiasNieuwenhuisen, 2019-10-02:
I fixed a similar issue in the Octomap RViz Plugin a while ago, probably the patch can also be applied here, but I currently have no time to investigate this. For reference, this is the PR solving the issue in the RViz Plugin:
https://github.com/OctoMap/octomap_rviz_plugins/pull/30

RMonica, 2019-10-06:
Thanks. I've ported your patch.
https://github.com/ros-planning/moveit/pull/1685

RMonica, 2020-02-10:
The pull request has been accepted.

I confirm this is fixed in master. I'm closing the issue.

closed 2020-02-10 by RMonica