moveit/moveit #1688: Docs and Debug Info for MoveIt Cartesian planner

xarthurx, 2019-10-09:
### Description

Hi, we are working on a project that needs a lot of Cartesian planning. But the results given by MoveIt is often not that ideal and we cannot find too much info about the default MoveIt Cartesian planner.

So, help needed from the community here...

Is there any documentation about the Cartesian planner used in MoveIt by default?
Besides the percentage of the given trajectory followed and the waypoints. What other info can we get for debugging? (e.g. if it is a local min, or reachability issue, or collision, etc.)
Thanks.

### Your environment
* ROS Distro: [IKinetic]
* OS Version: e.g. Ubuntu 16.04

gavanderhoorn, 2019-10-09:
This would appear to be a cross-post of [Docs and Debug Info for MoveIt Cartesian planner](https://answers.ros.org/question/334955) on ROS Answers.

xarthurx, 2019-10-09:
Yes, but neither got an answer.

felixvd, 2019-10-11:
The Cartesian planner in MoveIt does not work like the other planners - it simply generates waypoints along a line in Cartesian space and tries to find a robot pose from IK for each waypoint. The trajectory is returned up to the first waypoint where A) the IK returns no feasible result (out of the robot's reach), 2) a collision is detected or 3) the joints flip or move around too much between the waypoints (violation of the jump_threshold). The latter would happen most frequently near singularities, but can also problematic when you cross regions where the IK returns "flipped" . You can read the details in [cartesian_interpolator.cpp](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/src/cartesian_interpolator.cpp).

There has been discussion about changing the behavior to be more similar to the other planners, implementing [Descartes](http://wiki.ros.org/descartes) and other improvements. I believe the most recent ones are in #797 and #453.

If your project uses this feature a lot and you need extra debugging, please consider discussing it here and filing a PR if you find that anything is lacking.

felixvd, 2021-04-28:
I think there are plenty of ways to make debugging planning requests easier (e.g. publishing the goal pose and the expected robot pose at the goal to rviz) but that is an open-ended usability issue, so I'll close this. Feel free to reopen or discuss it further if there is something concrete. PRs are welcome.

closed 2021-04-28 by felixvd