moveit/moveit #1691: Xacro argument not passed to .srdf.xacro file

rickstaa, 2019-10-13:
### Description

I am trying to control the Panda Emika Franka robot using moveit. Since I neede to control relative to the gripper centre I added an extra virtual joint (meaning no collisions and visuals). To be able to switch this joint on and of I added the following `xacro:if` statement to the `panda_arm_hand.srdf.xacro`:

```
  <!-- Initialize gripper_center properties -->
  <xacro:arg name="gripper_center" default="true" />
  <xacro:property name="gripper_center" value="$(arg gripper_center)"/>

  <xacro:if value="${gripper_center}">    <!-- When gripper_center==True -->
    <group name="panda_arm_hand">
      <chain base_link="panda_link0" tip_link="panda_gripper_center" />
    </group>
  </xacro:if>
  <xacro:unless value="${gripper_center}">    <!-- When gripper_center==False -->
    <group name="panda_arm_hand">
      <group name="panda_hand" />
      <group name="panda_arm" />
    </group>
  </xacro:unless>

```

Unfortunately, when using the `moveit_setup_assistant` the xacro argument `gripper_center:=true` is only passed to the `panda_arm_hand.urdf.xacro` file while the `panda_arm_hand.srdf.xacro` file keeps using the default value I have set.

# Code
The `panda_moveit_config` branch I use can be found [here](https://github.com/rickstaa/panda_moveit_config/tree/melodic-devel-panda_autograsp). The accompanying `franka_ros` repository can be found [here](https://github.com/rickstaa/franka_ros/tree/melodic-devel-panda_autograsp).

### Your environment
* **ROS Distro:** Melodic
* **OS Version:** Ubuntu 18.04
* **Source:** v1.0.2

### Steps to reproduce
1. Clone my `panda_moveit_config` and `franka_ros` branches.
2. Build the catkin workspace.
3. Open the setup_assistant.
4. Select the `panda_moveit_config` in the setup assistant.
5. Set the xacro arguments to `center_gripper:=true`

### Expected behaviour
In the move_groups tab you should see the following:

[image]

### Actual behaviour
Now instead you see the following:

[image]

### Backtrace or Console output
[console log](https://gist.github.com/rickstaa/8f56d583818857d5506f5c7103bdc454)

rhaschke, 2019-10-13:
The xacro args that you can provide in MSA are intended for use with the urdf only.
If we want to have args for the SRDF file too, this actually requires a new, potentially different set of arguments.
To solve your issue, you can adapt the launch file `planning_context.launch` and insert your additional arguments there.

rickstaa, 2019-10-13:
@rhaschke Thanks for your fast answer. Feel free to close or add as a feature request.

closed 2019-10-18 by rickstaa