moveit/moveit #1696: steep changes in joint values after planning

ThanasisTs, 2019-10-17:
I am trying to make a UR3 move through waypoints using the `computeCartesianPath()` function. The returned value of the function is 1, which means that the whole planned trajectory will be executed. However, the visualization of the trajectory in Rviz shows steep changes in the configuration space, which results in undesired behaviour. The following gif shows the trajectory. I would like to know what might be the cause of this behaviour and why the planner considers such a plan as valid. Thanks in advance and please let me know if there is any additional information I could provide.
[image]

gavanderhoorn, 2019-10-18:
To be blunt: there is no "planning". Or at least not when considering the actual planning pipeline of MoveIt and comparing it to the functionality you're using here.

`computeCartesianPath(..)` essentially linearly interpolates between the poses you provide it, then uses IK to map that all to joint poses.

What you're seeing is typically the result of either:

 - a too high value set for `jump_factor`: this tells the interpolator that such discontinuous changes are ok
 - a misconfigured or misbehaving IK solver: if the only solution the solver returns is one which is far away from the previous one (and `jump_thresold` doesn't prevent it) then the interpolator will accept it and move on to the next point

If you're using Trac IK for instance, you should configure it for `Distance` solve type, not `Speed` (which is the default).

It's seen quite some improvements and refactors (#1293, #1149, etc), but `computeCartesianPath(..)` is not a true "planner", so it can do some strange things sometimes.

mamoll, 2019-10-18:
A side note, this kind of constrained planning _can_ be done correctly. We have implemented this in OMPL (see https://vimeo.com/261052837 for some videos of demos included with OMPL), but MoveIt's OMPL interface has yet to be updated. This might take some effort, so no easy fix for this particular issue.

v4hn, 2019-10-18:
The issue here is simply a high jump factor (or absolute jump thresholds, whatever they use), as @gavanderhoorn wrote. If you do not limit the maximum joint space distance between consecutive runs, big jumps can occur.
I'll close this issue as this is a standard issue that pops up regularly and has a clear answer (though no perfect solution).

@mamoll Since we have the additional support for the `reference_trajectories` field in `moveit_msgs/MotionPlanRequest`, we could actually specify Cartesian paths for planners.
Do you propose to write the glue code? :sunglasses:

closed 2019-10-18 by v4hn

ThanasisTs, 2019-10-18:
Thank you all for your helpful answers.

mamoll, 2019-10-18:
> @mamoll Since we have the additional support for the reference_trajectories field in moveit_msgs/MotionPlanRequest, we could actually specify Cartesian paths for planners.
Do you propose to write the glue code? 😎

Hmm, @zkingston what do you think?