moveit/moveit #1698: Rviz gets a SIGSEGV when ”Allow Approximate IK Solutions” is used.

smorita-esol, 2019-10-18:
# Description

The Rviz gets a SIGSEGV when I enabled the option ”Allow Approximate IK Solutions”.
We can reproduce this even with the robot model used in the quick start below.

http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html

# Your environment

* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binaries, versions: 1.0.2

# Steps to reproduce

Flow the instructions in the [quick start](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html), enabling the ”Allow Approximate IK Solutions” in the tab "Planning".
The Rviz will get a SIGSEGV after you drag the interactive marker (blue sphere) too much (about 30 times).

# Expected behaviour

The Rviz will never get any SIGSEGV by dragging the interactive marker.

# Actual behaviour

The Rviz gets a SIGSEGV.

# Backtrace or Console output

The backtrace caught by GDB when the phenomenum above occured.
```
#0  0x00007fcfb96e3b6a in fcl::HierarchyTree<fcl::AABB>::bottomup(__gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>\
*> > >, __gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >) () at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#1  0x00007fcfb96e4bbd in fcl::HierarchyTree<fcl::AABB>::topdown_0(__gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB\
>*> > >, __gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >) () at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#2  0x00007fcfb96e4b68 in fcl::HierarchyTree<fcl::AABB>::topdown_0(__gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB\
>*> > >, __gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >) () at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#3  0x00007fcfb96e4b68 in fcl::HierarchyTree<fcl::AABB>::topdown_0(__gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB\
>*> > >, __gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >) () at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#4  0x00007fcfb96e4e11 in fcl::HierarchyTree<fcl::AABB>::init_0(std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> >&) () at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#5  0x00007fcfb96e2085 in fcl::DynamicAABBTreeCollisionManager::registerObjects(std::vector<fcl::CollisionObject*, std::allocator<fcl::CollisionObject*> > const&) ()
    at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#6  0x00007fcfbb461d9e in collision_detection::FCLObject::registerTo(fcl::BroadPhaseCollisionManager*) () at /opt/ros/melodic/lib/libmoveit_collision_detection_fcl.so.1.0.2
#7  0x00007fcfbb4734b1 in collision_detection::CollisionRobotFCL::checkSelfCollisionHelper(collision_detection::CollisionRequest const&, collision_detection::CollisionResult&, moveit::core::RobotState \
const&, collision_detection::AllowedCollisionMatrix const*) const () at /opt/ros/melodic/lib/libmoveit_collision_detection_fcl.so.1.0.2
#8  0x00007fcfc20494da in planning_scene::PlanningScene::isStateColliding(moveit::core::RobotState const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) \
const () at /opt/ros/melodic/lib/libmoveit_planning_scene.so.1.0.2
#9  0x00007fcfc3d98aac in moveit_rviz_plugin::MotionPlanningDisplay::isIKSolutionCollisionFree(moveit::core::RobotState*, moveit::core::JointModelGroup const*, double const*) const ()
    at /opt/ros/melodic/lib/libmoveit_motion_planning_rviz_plugin_core.so.1.0.2
#10 0x00007fcfc293f4a2 in  () at /opt/ros/melodic/lib/libmoveit_robot_state.so.1.0.2
#11 0x00007fcfb7f09e79 in kdl_kinematics_plugin::KDLKinematicsPlugin::searchPositionIK(geometry_msgs::Pose_<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, double, s\
td::vector<double, std::allocator<double> >&, boost::function<void (geometry_msgs::Pose_<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, moveit_msgs::MoveItErrorCode\
s_<std::allocator<void> >&)> const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&, std::vector<double, std::allocator<double> > const&, kinematics::KinematicsQueryOptions const&) const ()
    at /opt/ros/melodic/lib//libmoveit_kdl_kinematics_plugin.so
#12 0x00007fcfb8b55c72 in kinematics::KinematicsBase::searchPositionIK(std::vector<geometry_msgs::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::Pose_<std::allocator<void> > > > const&, st\
d::vector<double, std::allocator<double> > const&, double, std::vector<double, std::allocator<double> > const&, std::vector<double, std::allocator<double> >&, boost::function<void (geometry_msgs::Pose_\
<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&)> const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void>\
 >&, kinematics::KinematicsQueryOptions const&, moveit::core::RobotState const*) const () at /opt/ros/melodic/lib/libmoveit_kinematics_base.so.1.0.2
#13 0x00007fcfc2948fde in moveit::core::RobotState::setFromIK(moveit::core::JointModelGroup const*, std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1\
, 0> > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char\
> > > > const&, std::vector<std::vector<double, std::allocator<double> >, std::allocator<std::vector<double, std::allocator<double> > > > const&, double, boost::function<bool (moveit::core::RobotState*\
, moveit::core::JointModelGroup const*, double const*)> const&, kinematics::KinematicsQueryOptions const&) () at /opt/ros/melodic/lib/libmoveit_robot_state.so.1.0.2
#14 0x00007fcfc294a3cc in moveit::core::RobotState::setFromIK(moveit::core::JointModelGroup const*, Eigen::Transform<double, 3, 1, 0> const&, std::__cxx11::basic_string<char, std::char_traits<char>, st\
d::allocator<char> > const&, std::vector<double, std::allocator<double> > const&, double, boost::function<bool (moveit::core::RobotState*, moveit::core::JointModelGroup const*, double const*)> const&, \
kinematics::KinematicsQueryOptions const&) () at /opt/ros/melodic/lib/libmoveit_robot_state.so.1.0.2
#15 0x00007fcfc294a7bd in moveit::core::RobotState::setFromIK(moveit::core::JointModelGroup const*, geometry_msgs::Pose_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits\
<char>, std::allocator<char> > const&, double, boost::function<bool (moveit::core::RobotState*, moveit::core::JointModelGroup const*, double const*)> const&, kinematics::KinematicsQueryOptions const&) \
()
    at /opt/ros/melodic/lib/libmoveit_robot_state.so.1.0.2
#16 0x00007fcfc34d9f83 in robot_interaction::KinematicOptions::setStateFromIK(moveit::core::RobotState&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__c\
xx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, geometry_msgs::Pose_<std::allocator<void> > const&) const () at /opt/ros/melodic/lib/libmoveit_robot_interaction.so.1.0.2
#17 0x00007fcfc34ec7c8 in robot_interaction::InteractionHandler::updateStateEndEffector(moveit::core::RobotState*, robot_interaction::EndEffectorInteraction const*, geometry_msgs::Pose_<std::allocator<\
void> > const*, boost::function<void (robot_interaction::InteractionHandler*)>*) () at /opt/ros/melodic/lib/libmoveit_robot_interaction.so.1.0.2
#18 0x00007fcfc34e6dae in robot_interaction::LockedRobotState::modifyState(boost::function<void (moveit::core::RobotState*)> const&) () at /opt/ros/melodic/lib/libmoveit_robot_interaction.so.1.0.2
#19 0x00007fcfc34ed0f9 in robot_interaction::InteractionHandler::handleEndEffector(robot_interaction::EndEffectorInteraction const&, boost::shared_ptr<visualization_msgs::InteractiveMarkerFeedback_<std\
::allocator<void> > const> const&) () at /opt/ros/melodic/lib/libmoveit_robot_interaction.so.1.0.2
#20 0x00007fcfc34f3b15 in robot_interaction::RobotInteraction::processingThread() () at /opt/ros/melodic/lib/libmoveit_robot_interaction.so.1.0.2
#21 0x00007fd04a087bcd in  () at /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#22 0x00007fd049e5e6db in start_thread (arg=0x7fcfb4a31700) at pthread_create.c:463
#23 0x00007fd04aa5c88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
````

rhaschke, 2019-10-18:
Is this related to the option ”Allow Approximate IK Solutions” or to the fast? / often? dragging of the mouse?

smorita-esol, 2019-10-21:
> Is this related to the option ”Allow Approximate IK Solutions” or to the fast? / often? dragging of the mouse?

It relates to the option ”Allow Approximate IK Solutions”. With the option, the rviz gets a SIGSEGV after I dragged the interactive marker about 30 times by mouse.

nick-paul, 2019-12-23:
I am also having this issue with a custom robot. Rviz crashes with a segfault when "Allow Approximate IK Solutions" is checked and I move the marker around a bit. My backtrace is below (*essentially the same as the one in the original comment*).

```
#0  0x00007fff64c2eb6a in fcl::HierarchyTree<fcl::AABB>::bottomup(__gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >, __gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >) ()
    at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#1  0x00007fff64c2fbbd in fcl::HierarchyTree<fcl::AABB>::topdown_0(__gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >, __gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >) ()
    at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#2  0x00007fff64c2fb7a in fcl::HierarchyTree<fcl::AABB>::topdown_0(__gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >, __gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >) ()
    at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#3  0x00007fff64c2fb68 in fcl::HierarchyTree<fcl::AABB>::topdown_0(__gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >, __gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >) ()
    at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#4  0x00007fff64c2fb68 in fcl::HierarchyTree<fcl::AABB>::topdown_0(__gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >, __gnu_cxx::__normal_iterator<fcl::NodeBase<fcl::AABB>**, std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> > >) ()
    at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#5  0x00007fff64c2fe11 in fcl::HierarchyTree<fcl::AABB>::init_0(std::vector<fcl::NodeBase<fcl::AABB>*, std::allocator<fcl::NodeBase<fcl::AABB>*> >&) ()
    at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#6  0x00007fff64c2d085 in fcl::DynamicAABBTreeCollisionManager::registerObjects(std::vector<fcl::CollisionObject*, std::allocator<fcl::CollisionObject*> > const&) ()
    at /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#7  0x00007fff669acd9e in collision_detection::FCLObject::registerTo(fcl::BroadPhaseCollisionManager*) () at /opt/ros/melodic/lib/libmoveit_collision_detection_fcl.so.1.0.2
#8  0x00007fff669be4b1 in collision_detection::CollisionRobotFCL::checkSelfCollisionHelper(collision_detection::CollisionRequest const&, collision_detection::CollisionResult&, moveit::core::RobotState const&, collision_detection::AllowedCollisionMatrix const*) const () at /opt/ros/melodic/lib/libmoveit_collision_detection_fcl.so.1.0.2
#9  0x00007fff856224da in planning_scene::PlanningScene::isStateColliding(moveit::core::RobotState const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) const () at /opt/ros/melodic/lib/libmoveit_planning_scene.so.1.0.2
#10 0x00007fff87597a9c in moveit_rviz_plugin::MotionPlanningDisplay::isIKSolutionCollisionFree(moveit::core::RobotState*, moveit::core::JointModelGroup const*, double const*) const ()
    at /opt/ros/melodic/lib/libmoveit_motion_planning_rviz_plugin_core.so.1.0.2
#11 0x00007fff85f144a2 in  () at /opt/ros/melodic/lib/libmoveit_robot_state.so.1.0.2
#12 0x00007fff62981e79 in kdl_kinematics_plugin::KDLKinematicsPlugin::searchPositionIK(geometry_msgs::Pose_<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, double, std::vector<double, std::allocator<double> >&, boost::function<void (geometry_msgs::Pose_<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&)> const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&, std::vector<double, std::allocator<double> > const&, kinematics::KinematicsQueryOptions const&) const () at /opt/ros/melodic/lib//libmoveit_kdl_kinematics_plugin.so
#13 0x00007fff640a0c72 in kinematics::KinematicsBase::searchPositionIK(std::vector<geometry_msgs::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::Pose_<std::allocator<void> > > > const&, std::vector<double, std::allocator<double> > const&, double, std::vector<double, std::allocator<double> > const&, std::vector<double, std::allocator<double> >&, boost::function<void (geometry_msgs::Pose_<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&)> const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&, kinematics::KinematicsQueryOptions const&, moveit::core::RobotState const*) const () at /opt/ros/melodic/lib/libmoveit_kinematics_base.so.1.0.2
#14 0x00007fff85f1dfde in moveit::core::RobotState::setFromIK(moveit::core::JointModelGroup const*, std::vector<Eigen::Transform<double, 3, 1, 0>, Eigen::aligned_allocator<Eigen::Transform<double, 3, 1, 0> > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char [... cut ...] ::allocator<void> > const> const&) () at /opt/ros/melodic/lib/libmoveit_robot_interaction.so.1.0.2
#21 0x00007fff86ac8b15 in robot_interaction::RobotInteraction::processingThread() () at /opt/ros/melodic/lib/libmoveit_robot_interaction.so.1.0.2
#22 0x00007ffff57c1bcd in  () at /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#23 0x00007ffff55986db in start_thread (arg=0x7fff4f97c700) at pthread_create.c:463
#24 0x00007ffff619688f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

rhaschke, 2020-01-04:
I can reproduce the issue - but only with ”Allow Approximate IK Solutions” checked. However, I have no idea why or how this flag should influence collision checking.
As the error only occurs after quite a while, I guess it's a race condition or maybe memory in FCL manager is accumulating. I don't have time to investigate further right now. Any help is very welcome.
I suggest to monitor memory usage and employ an address sanitizer to track down this issue.