moveit/moveit #1700: Specifying Mimic Joints

TomShkurti, 2019-10-22:
### Description

Mimic joints defined in the URDF are immobile (not represented as mimic joints in MoveIt), unless they are in an end effector.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build [fill in number]

### Steps to reproduce
Generated an SRDF from the following XACRO files:

https://github.com/cwru-davinci/uv_geometry/blob/moveit_handoff/xacro/psm_tool_sca.xacro
https://github.com/cwru-davinci/uv_geometry/blob/moveit_handoff/xacro/psm_base.xacro

### Expected behaviour
All joints with the URDF 'mimic' tag should move alongside the powered joints they are mimicking.

### Actual behaviour
Only the joints in psm_tool_sca (those related to the end effector) move, and those in psm_base are immobile.

rhaschke, 2019-10-22:
Did you selected the correct move group? If only joints from the end-effector group move, this looks like you have the end-effector group selected, but not the arm group?

saikishor, 2020-12-23:
Did you find the solution?. I am having this issue now with a robot which has a couple of mimic joints

rhaschke, 2020-12-24:
@saikishor, could you become more precise: Are mimic joints considered for:
- planning, i.e. the plan preview correctly shows their motion?
- fake execution
- real execution?

saikishor, 2020-12-24:
@rhaschke I had issues with the planning itself.  I will give it a try again post Xmas and get back to you with the proper log.

Thanks for such an imminent response

saikishor, 2021-06-21:
@rhaschke I have an issue, the rviz GUI doesn't consider the mimic joints when looking for a random valid goal state. Any idea, what could be the issue. I have added my mimic joints in the Passive Joints list, but it doesn't seem to do the job.

v4hn, 2021-06-21:
I can't reproduce this new issue with `moveit_resources`'s panda_moveit_config which has a mimic joint in the end-effector.
If you think you found a bug please provide a minimal example to reproduce the issue.

saikishor, 2021-06-21:
@v4hn In our case this happens with the mimic joints at a place other than the end-effector. Mainly linked to the joints that move. Moreover, as the end goal state in the rviz window is inappropriate we are not sure if the mimic configuration is taken into account for collision checking and the planning.

v4hn, 2021-06-21:
I'm sorry about you bad experience and in theory thank you for the bug report, but you leave us with nothing but your word that this problem actually exists and nobody is going to dig into the code without an actual error to debug.
If you want to see this fixed, please provide a minimal example in terms of a urdf and a corresponding moveit config that triggers the problem.

saikishor, 2021-06-21:
Sure. I'll try to provide one soon. Sorry for the trouble.
Thank you,