moveit/moveit #1701: demo pick.py fails

johnbmcd, 2019-10-23:
### Description

Running of the pick.py demo fails due to use of deprecated api. Running the demo results in the following error: 
`[ INFO] [1571788969.389385195]: Loading robot model 'panda'...
Traceback (most recent call last):
  File "/home/john/code/cs422/moveit_ws/src/moveit/moveit_commander/demos/pick.py", line 77, in <module>
    robot.right_arm.pick("part")
  File "/home/john/code/cs422/moveit_ws/src/moveit/moveit_commander/src/moveit_commander/robot.py", line 306, in __getattr__
    return object.__getattribute__(self, name)
AttributeError: 'RobotCommander' object has no attribute 'right_arm'
`

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source
* melodic-devel

### Steps to reproduce
run the demo.launch from https://github.com/ros-planning/panda_moveit_config.git
the `rosrun moveit_commander pick.py`

### Expected behaviour
pick plan should execute

### Actual behaviour
see above error

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2019-10-23:
Thanks for reporting this issue. `right_arm` refers to the PR2 robot, which has a left and right arm (and corresponding move groups). If you replace this group name with `panda_arm` it works - in principle. However, the planning scene layout is not suitable for the smaller Panda robot.
I suggest looking into the [pick-and-place tutorials](https://ros-planning.github.io/moveit_tutorials/doc/pick_place/pick_place_tutorial.html), which are specifically designed for the Panda.

closed 2019-10-23 by rhaschke

johnbmcd, 2019-10-23:
Hi @rhaschke, thanks for the suggestion. I have reviewed the pick-and-place tutorials at length and after much effort to try to recreate the cpp code in python, I believe there is either a bug in the python implementation of the move_group pick function or some step in the process which is not well documented (e.g. in setting up the collision objects). 

In order to test this I have taken the pick_place_tutorial.cpp and reduced it to just the pick step. I have then (to the best of my knowledge) reimplemented this exact functionality in python using the moveit python api (i.e. both have the same planning scene, grasp specs, etc.). Both files are included in the following zip file:
[pick_tutorial.zip](https://github.com/ros-planning/moveit/files/3764932/pick_tutorial.zip)

As far as I understand it, both of these should result in identical behaviour, however, although the cpp implementation performs correctly every time, the python implementation always fails with the following error message:
```
[ INFO] [1571872052.627170581]: Loading robot model 'panda'...
[ INFO] [1571872053.834468268]: Ready to take commands for planning group panda_arm.
[ WARN] [1571872059.106285895]: Fail: ABORTED: No motion plan found. No execution attempted.
```

The panda_moveit_config demo.launch outputs the following:
```
...
You can start planning now!

[ INFO] [1571872049.662848153]: Loading robot model 'panda'...
[ INFO] [1571872049.989410519]: Loading robot model 'panda'...
[ WARN] [1571872050.018825851]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_JMCD_L7400_16266_721861437210336842/panda_arm/kinematics_solver_attempts' from your configuration.
[ INFO] [1571872050.080130570]: Starting planning scene monitor
[ INFO] [1571872050.082205605]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1571872058.899590539]: Planning attempt 1 of at most 1
[ INFO] [1571872058.903605745]: Added plan for pipeline 'pick'. Queue is now of size 1
[ INFO] [1571872059.004795082]: Manipulation plan 0 failed at stage 'reachable & valid pose filter' on thread 2
[ WARN] [1571872059.004980820]: All supplied grasps failed. Retrying last grasp in verbose mode.
[ INFO] [1571872059.005084368]: Re-added last failed plan for pipeline 'pick'. Queue is now of size 1
[ INFO] [1571872059.055315335]: IK failed
[ INFO] [1571872059.105423256]: IK failed
[ INFO] [1571872059.105537733]: Sampler failed to produce a state
[ INFO] [1571872059.105568832]: Manipulation plan 0 failed at stage 'reachable & valid pose filter' on thread 1
[ INFO] [1571872059.105792647]: Pickup planning completed after 0.105249 seconds
[ INFO] [1571872069.454597341]: Stopping planning scene monitor

```

My apologies if I am incorrect here, and this is not a bug. If so, I can always redirect to answers.ros.org -- however a representative example of how the api should be used would be helpful as part of the docs.

rhaschke, 2019-10-24:
Unfortunately, I don't have time to look into the details of your issue right now. 
As you want to reproduce the pick-and-place demo in python, you might want to have a look at https://github.com/ros-planning/moveit_tutorials/pull/406.
@mlautman, maybe it is a good idea to include a python version of the pick-and-place demo in the tutorials as people are looking for this...

felixvd, 2019-10-24:
@johnbmcd Am I understanding right that the cpp version of the tutorial works, but when you write it in Python, it doesn't? It might have something to do with the number of planning attempts or pipelines that are launched by default (I don't think it should only be 1; I had more success with at least 5-10), but since the IK fails to produce a valid pose, it seems likely that there is a collision somehow. Are the collision objects set up the same way as in the original? Is the gripper open/close position set correctly? Are the orientations correct? Does the scene look right in Rviz?

I'd reopen, it seems like the issue isn't as simple as renaming the move group.

johnbmcd, 2019-10-24:
@felixvd, initially my answer to this would have been yes!, however having reviewed the example provided in the pull request provided by @rhaschke I identified that I had made an error in one of the frame_id's in the grasp pose. I can confirm that this is now working based on @rickstaa's example. Although I would have spotted this quicker with an example in python I do appreciate @mlautman arguments regarding maintaining two versions of the same example, etc. In trying to port the example I did come across a number of people also trying this so perhaps a link to @rickstaa's example somewhere in the docs would be helpful. Thanks for the help, and apologies for the diversion!