moveit/moveit #1707: trajectory planning through path constraints

ThanasisTs, 2019-10-31:
Hello,

I am trying to move the end point of an UR3 robotic manipulator through a series of waypoints. The approach I am using is that I specify the final point of the waypoints as the goal point and all the other waypoints as path constraints. Basically, I want to tell the planner the reach the goal point as long as it manages to pass through all the other points.

For that purpose, I am using the motion planning pipeline of Moveit! and specifically I create a MotionPlanRequest. For each waypoint used to create a path constraint, I create a sphere centered at that point and of radius which is experimentally derived so that each consecutive spheres overlap (so that no hole in the final swept volume exists). After that, I move the end effector to the first waypoint and i specify the final waypoint as the goal point. The problem is that when i call the planner, it seems that it doesn't take into the consideration the trajectory constraints that I specify and it simply moves the end effector to the goal point. I would like to know why the trajectory constraints are being omitted, since according to [http://docs.ros.org/melodic/api/moveit_msgs/html/msg/MotionPlanRequest.html](url) the _trajectory_constraints_ field encapsulates all the constraints the resulting trajectory should satisfy. Or is it that there is something that I have completely misunderstood about the functionality of path constraints? Thanks in advance! Below the code that is described above is shown.

`int num_of_points = waypoints.points.size();

  geometry_msgs::PoseStamped pose;
  pose.header.frame_id = "world";
  
// //
  // // Move to the first point
  // //

  pose.pose.position.x = waypoints.points[0].x;
  pose.pose.position.y = waypoints.points[0].y;
  pose.pose.position.z = waypoints.points[0].z;
  pose.pose.orientation.x = -0.001;
  pose.pose.orientation.y = 0.685;
  pose.pose.orientation.z = -0.001;
  pose.pose.orientation.w = 0.729;

  std::vector<double> tolerance_pose(3, 0.01);
  std::vector<double> tolerance_angle(3, 0.01);

  req.group_name = "arm";
  moveit_msgs::Constraints init_goal = kinematic_constraints::constructGoalConstraints("ee_link", pose, tolerance_pose, tolerance_angle);
  req.goal_constraints.push_back(init_goal);
  req.allowed_planning_time = 100.0;  

  // // Now, call the pipeline and check whether planning was successful.
  planning_pipeline->generatePlan(planning_scene, req, res);
  /* Check that the planning was successful */
  if (res.error_code_.val != res.error_code_.SUCCESS){
    ROS_ERROR("Could not compute plan successfully");
    exit (-1);
  }
  
  moveit_msgs::DisplayTrajectory display_trajectory;
  moveit_msgs::MotionPlanResponse response;
  res.getMessage(response);

  display_trajectory.trajectory_start = response.trajectory_start;
  display_trajectory.trajectory.push_back(response.trajectory);
  display_publisher.publish(display_trajectory);

  moveit::planning_interface::MoveGroupInterface mg("arm");
  moveit::planning_interface::MoveGroupInterface::Plan myplan;
  myplan.trajectory_ = response.trajectory;
  mg.execute(myplan);

  //
  // Set the last point as goal constraint
  //
  pose.pose.position.x = waypoints.points[num_of_points-1].x;
  pose.pose.position.y = waypoints.points[num_of_points-1].y;
  pose.pose.position.z = waypoints.points[num_of_points-1].z;

  pose.pose.orientation.x = -0.001;
  pose.pose.orientation.y = 0.685;
  pose.pose.orientation.z = -0.001;
  pose.pose.orientation.w = 0.729;
  
  moveit_msgs::Constraints end_goal = kinematic_constraints::constructGoalConstraints("ee_link", pose, tolerance_pose, tolerance_angle);
  
  req.goal_constraints.clear();
  req.goal_constraints.push_back(end_goal);
  
  //
  // Set the interpolated points as position constraints and create a respective constraint region
  //
  geometry_msgs::PointStamped point;
  point.header.frame_id = "world";
  moveit_msgs::Constraints point_constraint;
  
  for (short int i=0; i<num_of_points; i++){
    point.point.x = waypoints.points[i].x;
    point.point.y = waypoints.points[i].y;
    point.point.z = waypoints.points[i].z;
  
    point_constraint = kinematic_constraints::constructGoalConstraints("ee_link", point, 0.05);
    req.trajectory_constraints.constraints.push_back(point_constraint);
  }

  ROS_INFO("Created the path constraints...");

// Call the pipeline
  planning_pipeline->generatePlan(planning_scene, req, res);
  
//Check that the planning was successful
  if (res.error_code_.val != res.error_code_.SUCCESS)
  {
    ROS_ERROR("Could not compute plan successfully");
    exit(-1);
  }

  display_publisher = node_handle.advertise<moveit_msgs::DisplayTrajectory>("/move_group/display_planned_path", 1, true);

  res.getMessage(response);

  display_trajectory.trajectory_start = response.trajectory_start;
  display_trajectory.trajectory.push_back(response.trajectory);
  display_publisher.publish(display_trajectory);

  moveit::planning_interface::MoveGroupInterface::Plan myplan2;
  myplan2.trajectory_ = response.trajectory;
  mg.execute(myplan2);`

gokul-gokz, 2019-10-31:
I think , the best way to solve your waypoint navigation problem is to use either [cartesian path planner](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html) or the [Descarte path planner](https://github.com/PickNikRobotics/descartes_capability) in the moveit rather than imposing path constraints while planning for a single goal. As far as I know, path constraints are imposed generally to maintain the orientation or to maintain the configuration within certain bounds.

ThanasisTs, 2019-11-01:
The problem with the cartesian path planner is that it requires to define the orientation of the end effector for every waypoint in the trajectory, which is very restrictive. But I will check the Descarte path planner.. Thanks for your quick response.

gokul-gokz, 2019-11-01:
Yes. That would be a good option as Descarte planner allows you to define waypoints without having Z constraint of last joint.

v4hn, 2019-11-04:
The `MotionPlanRequest` API allows to set the `trajectory_constraints` field, but a planner plugin actually has to take them into account.
The default OMPL interface does not do that.

Incidentally, I'm not aware of a planner plugin that does that at the moment -
also because the semantics of these constraints are not very well-defined.

You could, of course, plan a series of motions from waypoint to waypoint, but you probably dismissed this as it likely produces very jerky paths.
You could smooth them by blending the trajectories, but that will make your end effector miss the exact waypoints.

Also, if you insist that you don't want to specify the orientation of the waypoints,
the problem gets much more complicated and you will need some extended mechanism to generate smooth paths.

Minor side note: if you try to use Descartes, make sure to use [the analytical IK solver](https://github.com/ros-industrial/universal_robot/tree/kinetic-devel/ur_kinematics) as Descartes relies on solving a lot of IK requests.

closed 2019-11-04 by v4hn

samanfahandezh, 2023-05-18:
Hi @v4hn, I was wondering if there is any progress on this front? The planner _Default OMPL_ does not take into account constraints set into the `MotionPlanRequest` by planning Pipeline. I'm trying to set a goal orientation constraint for an end-effector (ee orientation should be normal to a surface, so setting tol. for two axes small and for the third one large). As far as I know setting goal constraints (not path or trajectory constraints) is only possible by the planning pipeline API, so then I have this problem that the planner doesn't pick up the constraints set by req while planning.