moveit/moveit #1708: Configurable time parametrization method for default cartesian path planner in moveit

gokul-gokz, 2019-10-31:
### Description

Moveit offers three different time parametrization techniques which can be changed as per the user requirement from planning adapter. But the Cartesian path planner of the moveit always uses the IterativeparabolicParameterization technique as it is hard coded. This is already reported in the  [issue](https://answers.ros.org/question/326511/using-iterativesplineparameterization-with-computecartesianpath/) . I think, it would be best if it's configurable. I am happy to make the changes and do a PR. But, I want to know what would be the best way to do it.

Method 1:
Making changes in the planning request adapter.

Method 2:
Modifying the cartesian path plan service to accept one more parameter to choose the time parametrization method.
I have already did this through this method. But , I am not sure whether is this the best way.

mamoll, 2019-11-03:
Since you already have an implementation of method 2, why not just submit it as a PR and mention method 1 as a possible alternative approach in the description?

gokul-gokz, 2019-11-04:
Sure. Will do it.