moveit/moveit #1711: Trajectory smoothness

janisa9, 2019-11-05:
### Description

Actually I don't know where to start and which package makes the issue, but I will try my best to express this issue as best as I can. In a few words I would describe this- trajectory produced by MoveIt! is not as smooth as it should be.

I am using UR5, with UrSim 3.9 version and the new Universal_Robots_ROS_Driver https://github.com/UniversalRobots/Universal_Robots_ROS_Driver , I think no problems in this package.

To show the issue I am moving robot between 2 predefined positions without any objects between them, using Move Group C++ Interface and plotting joint velocities. 

After going into details in several similar issues #416 #167 , I have tried all the Time Parameterization Algorithms and different motions planners. There was no difference between OMPL (RRTConnect) and STOMP in terms of trajectory smoothness.  Here you can see the plots:

Using Time optimal trajectory generation
[image]

Using Iterative Parabolic Time Parameterization
[image]

Using Iterative Spline Parameterization
[image]

As you can see TOTG seems to give the best results in trajectory, but anyway there are some velocity spikes which causes jiggle and sometimes trigger safety on the real robot. As in the next picture

[image]

I have tried this on two separate pcs both running on kinetic, I've reinstalled and checked most of the packages, but could not find the solution. Maby I am missing something on this? 

If any additional information is needed please inform me.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build Melodic version on Kinetic
* melodic-devel 5344c09 [origin/melodic-devel] depth_image_octomap_updater: reset depth transfer function to standard values (#1661)

gavanderhoorn, 2019-11-05:
Although I recognise the issue with the various time parameterisation algorithms, the fact that you're looking at the `joint_states` topic implies that this is state coming back from the robot, after execution.

As the `joint_trajectory_controller` is the entity that ultimately resamples the trajectory that MoveIt sends it, I would suggest to take a look there.

The problem could be due to resampling artefacts (the JTC must resample, as the `JointTrajectory` message is much too sparse (in time) to be executed on the UR controller directly).

I would also suggest to make sure you are running a real-time kernel -- as recommended by the setup instructions of the new driver -- as hickups in the execution of the JTC could also cause this (so called 'catch up' behaviour).

janisa9, 2019-11-05:
Thank you for the quick response.

Yes indeed, I am not running a real-time kernel. In fact, I am already preparing another work-station with a clean install, where I will use a real-time kernel. Will make tests and see if it resolves "catch up" behaviour.

In the meantime to ensure that this is the problem would comparison between trajectory published on `/scaled_pos_traj_controller/follow_joint_trajectory/goal` and state coming back from the robot on `joint_state` topic be able to prove this on some level?

gavanderhoorn, 2019-11-05:
Comparison between the incoming trajectory and the way it is eventually executed would certainly be a good thing to do, yes.

I would perhaps also see what the `state` of the controller is. This is published by the JTC on a topic.

That would show you the difference between what the controller outputs and what the robot actually executes.

janisa9, 2019-11-06:
Yes, I think this pretty much shows the problem and using a real-time kernel should solve this. Thank you! 

[image]

Anyway, do you have ant idea why Iterative Spline Parameterization makes this weird trajectory?

janisa9, 2019-11-11:
Installed fresh ubuntu 18.04 with real-time kernel and ROS-melodic, the problem is gone. So probably this can be closed.

But I am still curious about Iterative Parabolic Time Parameterization and Iterative Spline Parameterization why the trajectory differs so much from TOTG and the fact why IPTP is used by default in MoveIt!?

v4hn, 2019-11-11:
> why IPTP is used by default in MoveIt!?

For historic reasons. You are free to create a pull-request to change the default.

TOTG can move the waypoints returned by sampling. This is not a very nice characteristic with the current pipeline and some additional collision checking would have to be added.

ISP was meant to be a sane replacement for IPTP, but somehow didn't make it to the default yet.

IPTP usually isn't the best, but was claimed to be the fastest at some point (never benchmarked myself and I'm not convinced the difference, if any, is relevant these days).

closed 2019-11-11 by v4hn

felixvd, 2021-01-28:
Related: https://github.com/ros-planning/moveit/issues/2183