moveit/moveit #1713: Unable to move with setPoseTarget for different eef link

azhural, 2019-11-06:
### Description

Using setPoseTarget(pose, link) and then moving works only until a different link is used.
I can confirm that the new pose is reachable, as swapping the setPoseTarget/Move-Blocks, rebuilding and then restarting only that node will allow the robot to move to the specified pose. It will then fail to move to the previous pose.
Using the same link allows multiple moves.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary
* 1.0.2-1bionic.20191023...

### Steps to reproduce

```
  geometry_msgs::PoseStamped pose_1, pose_2;
  // fill in poses
  move_group.setStartStateToCurrentState();
  move_group.setPoseTarget(pose_1, "link_1");
  move_group.move();  // works fine

  move_group.setStartStateToCurrentState();
  move_group.setPoseTarget(pose_2, "link_2");
  move_group.move();  // fails
```

```
  geometry_msgs::PoseStamped pose_1, pose_2;
  // fill in poses, same as first example
  move_group.setStartStateToCurrentState();
  move_group.setPoseTarget(pose_2, "link_2");
  move_group.move();  // works fine

  move_group.setStartStateToCurrentState();
  move_group.setPoseTarget(pose_1, "link_1");
  move_group.move();  // fails
```

```
  geometry_msgs::PoseStamped pose_1, pose_2;
  // fill in poses
  move_group.setStartStateToCurrentState();
  move_group.setPoseTarget(pose_1, "link_1");
  move_group.move();  // works fine

  move_group.setStartStateToCurrentState();
  move_group.setPoseTarget(pose_2, "link_1");
  move_group.move();  // works fine
```

### Expected behaviour
Both moves are executed.

### Actual behaviour
Only first move is executed.

### Backtrace or Console output
`[ERROR] [1573047099.608346833, 870.895000000]: RRTConnect: Insufficient states in sampleable goal region`

v4hn, 2019-11-06:
Calling `setPoseTarget` with a different link does not change the link of the target.  Instead it adds *a second* constraint for the goal to have the new link at a specific pose *in addition* to the first target set.

Apparently, there is no kinematic solution for the two poses you request for the two links at the same time.

Call `clearPoseTargets()` to remove the previously added constraint.

v4hn, 2019-11-06:
BTW: it is not necessary to call `setStartStateToCurrentState()` before each request.
The current state is only filled in in the `move_group` node and the function only ensures that you forget previously set start states which you defined manually.

closed 2019-11-06 by v4hn

azhural, 2019-11-06:
Thanks. I was under the impression that a successful move would clear the target.