moveit/moveit #1714: better documentation for calling `MGI::setPoseTarget` with different links

v4hn, 2019-11-06:
Issues such as https://github.com/ros-planning/moveit/issues/1713 pop up from time to time.
It seems the API behavior is rather non-intuitive for most users, so we should review the documentation of the function and maybe add it somewhere more prominently.

felixvd, 2019-11-07:
Thanks for posting, I also had this on my Todo list.

v4hn, 2019-11-07:
Feel free to add lots of issues like this over the next two weeks for WMD!

shivaang12, 2019-11-13:
This might sound too naive but I think we should provide two different operations where one operation auto resets the target link after successful `move()` while other doesn't - targeting advance users.

felixvd, 2019-11-14:
I can't really see the use case and would argue in favor of a unified behavior. More options means more confusion and maintenance.

shivaang12, 2019-11-14:
Cool! I was looking for something to work on World Moveit Day!